espressif/idf-extra-components #29: Consider setting up dependabot to update submodule dependencies

igrr, 2022-03-19:
Dependabot has basic support for tracking and upgrading dependencies expressed using git submodules (https://docs.github.com/en/code-security/dependabot/dependabot-version-updates/configuration-options-for-the-dependabot.yml-file#package-ecosystem). Consider setting it up for this repository to get notified whenever a dependency can be upgraded.

Note that we might need to wait for or contribute to https://github.com/dependabot/dependabot-core/issues/1639 first, as currently dependabot will try to upgrade to the latest commit, not to the latest tag.

tore-espressif, 2022-03-25:
Done.

I don't have any experience with dependabot, so I'll keep this issue open for future evaluation/discussion

igrr, 2022-03-25:
Thanks @tore-espressif!

The PRs opened by dependabot have two problems now:

1. The PR updates the submodule to the latest commit, not to the latest tag (issue https://github.com/dependabot/dependabot-core/issues/1639 noted above).
2. Looks like the "publish test results" job is failing.

Still even in this form they are useful as a hint/reminder to us that some dependency might be outdated.

igrr, 2023-02-20:
This might be worth revisiting, especially since https://github.com/espressif/idf-extra-components/pull/146 adds two libraries which are known to sometimes have CVEs reported for them. It would be good to be able to get the new releases of these libraries published quickly.