espressif/idf-extra-components #74: coap: Upgrade libcoap to more recent commit, to fix blockwise GET issues

ncmiller, 2022-07-21:
I would like to request that the version of libcoap be bumped to a more recent commit.

With the current commit https://github.com/obgm/libcoap/tree/3aa11612c143c9734d72022720f33e12506f7a2c, I ran into issues related to blockwise GET transfers. After investigating, I realized that it was a bug that has since been fixed in libcoap. I don't recall the specifics of the bug, or which PR/commit fixed it, but I might be able to find out if that's needed.

For my use, I forked the `coap` component, and updated `libcoap` to https://github.com/obgm/libcoap/tree/50128de191d558593bc9a48a08eb04735f942e45. This fixed the blockwise GET issues. However, it's not ideal for me to maintain the fork, and it would preferable if it could be updated here in this repo.

mahavirj, 2022-07-25:
@ncmiller

> I ran into issues related to blockwise GET transfers. After investigating, I realized that it was a bug that has since been fixed in libcoap. I don't recall the specifics of the bug, or which PR/commit fixed it, but I might be able to find out if that's needed.

It will be good to understand specific commit that fixed this problem.

@mrdeep1 Would you know about this?

mrdeep1, 2022-07-25:
There has been a fair amount of work in this area since release 4.3.0, including functional enhancement.  An idea as to what the issue being run into was would be helpful here to pinpoint the PR that fixed the issue

As a side note, the current develop branch should be both source and binary compatible with the 4.3.0 branch.  It is now possible to build the client-only or server-only versions of libcoap (would need esp-idf specific build options to be set up) to reduce code sizes.

mahavirj, 2022-07-25:
@mrdeep1 When is next release planned for `libcoap`? Or would you recommend update to specific commit from `develop` branch?

mrdeep1, 2022-07-25:
There are some big items to go into libcoap before the next planned release, so this would not be soon.  The current version of develop appears to be stable and may be the best way to go.  I will check back with the owner of libcoap as to whether we can come up with some sort of interim tag in the short term to help indicate a library module change.

mrdeep1, 2022-07-25:
@mahavirj @ncmiller That suggests to me that you are not using unique tokens for each request.  This commit partially supports using the same (or 0 length, i.e empty) token.  A later commit that what you are using https://github.com/obgm/libcoap/pull/869 (you are using https://github.com/obgm/libcoap/pull/847) does a better job of this as well as warn you when you are trying to the use the same token as there still are opportunities to get responses confused as to what was the request when 2 requests use the same token.

In simple terms, coap_cancel_all_messages() was deleting any sendqueue entry for Token X.  Then, when the time came to take the next entry off the session's delayqueue (also with Token = X) (put there because NSTART >= 1 as you are using CON) it did not happen as the sendqueue had nothing on it at that point and so the next entry was never transmitted.

I'm looking to whether we can easily get a tagged version of develop updated in MbedTLS.

ncmiller, 2022-07-25:
> That suggests to me that you are not using unique tokens for each request. This commit partially supports using the same (or 0 length, i.e empty) token.

That is correct. I was under the impression that the same token needed to be used for each block, since some of the BLOCK2 examples in RFC 7959 used the same token. But after reading RFC 7959 closer, I don't think this is a requirement:

> As a general comment on tokens, there is no other
   mention of tokens in this document, as block-wise transfers handle
   tokens like any other CoAP exchange.  As usual, the client is free to
   choose tokens for each exchange as it likes.

mrdeep1, 2022-07-25:
[RFC9175 4. 1 Token Processing for Secure Request-Response Binding](https://www.rfc-editor.org/rfc/rfc9175.html#name-token-processing-for-secure) goes someway to clarify the reasons for the use of Unique tokens - which also should be done when doing Block2 requests for the next block.

Incidentally, libcoap can do all of the block handling should you wish - removing the requirement of the application knowing what to do.  The application just needs to call `coap_context_set_block_mode()` to set this up.  See [coap_block(3)](https://libcoap.net/doc/reference/develop/man_coap_block.html).

mrdeep1, 2022-07-27:
@ncmiller FYI  #75 has been raised to update the version of libcoap.

PR #75 (2022-07-28): coap: Update libcoap library to commit 8fbe440

# Checklist

- [ ] Component contains License
- [ ] Component contains README.md
- [ ] Component contains idf_component.yml file with `url` field defined
- [ ] Component was added to [upload job](https://github.com/espressif/idf-extra-components/blob/master/.github/workflows/upload_component.yml#L18)
- [ ] Component was added to [build job](https://github.com/espressif/idf-extra-components/blob/master/test_app/CMakeLists.txt#L8)
- [ ] _Optional:_ Component contains unit tests
- [ ] CI passing

# Change description

Fixes #74

Provides new functio [... cut]
commit f1830684 (2022-07-28): coap: Update libcoap library to commit 8fbe440

Fixes #74

Provides new functionality to build Client Only or Server Only code.

Provides new functionality to include TCP (and hence TLS) support.

Removes the requirement for -Wno-warning
closed 2022-07-28 by mahavirj