espressif/idf-extra-components #106: Does the esp_jpeg component support SIMD?

modi12jin, 2022-11-14:
When I learned that SIMD instructions can speed up JPEG decoding, I wondered if the esp_jpeg component supports it?

igrr, 2023-01-25:
Not yet, but this is something we are planning to implement!

jason-mao, 2023-02-07:
The ADF  https://github.com/espressif/esp-adf-libs/blob/master/esp_codec/include/codec/esp_jpeg_dec/enc.h  has been optimized by S3 instructions.

tore-espressif, 2025-06-24:
The SIMD acceleration for JPEG decoding _and_ encoding is implemented in [esp_new_jpeg](https://components.espressif.com/components/espressif/esp_new_jpeg/) component. On ESP32-S3 it offers 3-4x performance boost over the old ROM implementation.

I'll close this issue as 'done'. Feel free to ask more questions here or in [esp_new_jpeg repository](https://github.com/espressif/esp-adf-libs/tree/35c37e1656db36dbf13b9aee8064d1b59f482f6c/esp_new_jpeg)

closed 2025-06-24 by tore-espressif

dizcza, 2025-12-28:
I confirm a steady x4 speedup with the new JPEG decoding library when scaling the image by half and x3 speedup with no scaling.

What I'd like to see in the new lib is decoding to gray support. Most image data analysis on esp32 is done to check for motion as with the PIR sensor...