espressif/idf-extra-components #116: tinyusb question

wuyuanyi135, 2022-12-18:
Hi, I noticed the TinyUSB library is scattered everywhere in the framework
* https://components.espressif.com/components/espressif/esp_tinyusb (Provides wrappers for some class but no ECM & RNDIS support)
* https://components.espressif.com/components/espressif/tinyusb (Seems the vanilla TinyUSB)
* https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/components/usb (With ECM & RNDIS)
And each library is maintaining their own features. I wonder whether there is any plan to merge them.

tore-espressif, 2022-12-19:
Hello @wuyuanyi135 , thank you for your interest in usb.

The first two packages you listed are long-term support of USB device on ESP SoCs. They are not going to be merged together. On the contrary, we split them for IDF v5. Some users prefer working with 'vanilla TinyUSB' while others prefer to use our additions (or wrappers). With this solution, the users can pick one of the two options.

TinyUSB code in esp-iot-solution servers as incubator for new USB device features. You can use the code if it works for your case, but there is limited support.

Eventually we'd like to make the long-term support TinyUSB feature-full, including the features that are already in esp-iot-solution.

wuyuanyi135, 2022-12-19:
@tore-espressif Thank you for the clarification. Currently TinyUSB serves as an important gap-filler for host-device communication when WiFi is not available. I really appreciate espressif's work to work on these libraries.

closed 2022-12-19 by wuyuanyi135