espressif/idf-extra-components #117: pre_encrypted_ota example fails in esp_encrypted_img_decrypt_end()

ashelman-ec, 2022-12-18:
I am running `examples/system/ota/pre_encrypted_ota/` with
ESP-IDF: `v5.0`
esp_encrypted_img: `v2.0.2`

Since this is just a test for now, I have not changed the rsa_key/private.pem file.

HTTPS connection to the OTA server succeeds. Originally the download was failing to complete; it seems the server was closing the connection before all data was transferred. Here is example log output (at DEBUG log level):
```
D (53877) HTTP_CLIENT: need_read=512, byte_to_read=512, rlen=512, ridx=512
D (53887) HTTP_CLIENT: http_on_body 512
D (53897) esp_https_ota: Written image length 590336
D (53897) pre_encrypted_ota_example: Image bytes read: 590336
D (53907) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=932336
D (53937) HTTP_CLIENT: need_read=1024, byte_to_read=512, rlen=-1, ridx=0
W (53937) HTTP_CLIENT: esp_transport_read returned:-1 and errno:128 
D (53937) HTTP_CLIENT: Data processed 589824 != Data specified in content length 932336
E (53947) esp_https_ota: data read -1, errno 128
D (53947) HTTP_CLIENT: Data processed 589824 != Data specified in content length 932336
E (53957) pre_encrypted_ota_example: Complete data was not received.
E (53967) pre_encrypted_ota_example: ESP_HTTPS_OTA upgrade failed
```

Setting "Enable partial HTTP download" in SDK Configuration editor solved that problem & the full download of the binary completes.

But then `esp_encrypted_img_decrypt_end()` fails with `ESP_FAIL` (I have added an error log to the example code):
```
    if (!esp_https_ota_is_complete_data_received(https_ota_handle)) {
        // the OTA image was not completely received and user can customise the response to this situation.
        ESP_LOGE(TAG, "Complete data was not received.");
    } else {
        err = esp_encrypted_img_decrypt_end(decrypt_handle);
        if (err != ESP_OK) {
-->         ESP_LOGE(TAG, "esp_encrypted_img_decrypt_end() returned %d = %s", err, esp_err_to_name(err));
            goto ota_end;
        }
```

Below is the end of the log output (at DEBUG log level).

**What is causing `esp_encrypted_img` to log the error `Invalid operation`?**

```
I (135817) esp-x509-crt-bundle: Certificate validated
D (136637) HTTP_CLIENT: Write header[4]: GET /ota/pre_encrypted_ota_secure.bin HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: myserver.net
Content-Length: 0
Range: bytes=918016-

D (136917) HTTP_CLIENT: on_message_begin
D (136917) HTTP_CLIENT: HEADER=Date:Sun, 18 Dec 2022 13:48:24 GMT
D (136917) HTTP_CLIENT: HEADER=Server:Apache
D (136917) HTTP_CLIENT: HEADER=Vary:IS_SUBREQ,User-Agent
D (136927) HTTP_CLIENT: HEADER=Upgrade:h2
D (136927) HTTP_CLIENT: HEADER=Connection:Upgrade
D (136927) HTTP_CLIENT: HEADER=Last-Modified:Sun, 18 Dec 2022 08:00:42 GMT
D (136937) HTTP_CLIENT: HEADER=ETag:"e39f0-5f0159970331c"
D (136947) HTTP_CLIENT: HEADER=Accept-Ranges:bytes
D (136947) HTTP_CLIENT: HEADER=Content-Length:14320
D (136957) HTTP_CLIENT: HEADER=Cache-Control:max-age=172800
D (136957) HTTP_CLIENT: HEADER=Expires:Tue, 20 Dec 2022 13:48:24 GMT
D (136967) HTTP_CLIENT: HEADER=Content-Range:bytes 918016-932335/932336
D (136977) HTTP_CLIENT: HEADER=Content-Type:application/octet-stream
D (136977) HTTP_CLIENT: http_on_headers_complete, status=206, offset=421, nread=421
D (136987) HTTP_CLIENT: content_length = 14320
D (136987) esp_https_ota: Connection start
D (136997) pre_encrypted_ota_example: Image bytes read: 918016
D (136997) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=14320
D (137037) HTTP_CLIENT: need_read=1024, byte_to_read=512, rlen=512, ridx=0
D (137037) HTTP_CLIENT: http_on_body 512
D (137037) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=14320
D (137047) transport_base: remain data in cache, need to read again
D (137047) HTTP_CLIENT: need_read=512, byte_to_read=512, rlen=512, ridx=512
D (137057) HTTP_CLIENT: http_on_body 512
D (137107) esp_https_ota: Written image length 919040
D (137107) pre_encrypted_ota_example: Image bytes read: 919040
D (137107) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=14320
D (137117) transport_base: remain data in cache, need to read again
( ... etc ...)
D (138007) HTTP_CLIENT: need_read=1024, byte_to_read=512, rlen=512, ridx=0
D (138017) HTTP_CLIENT: http_on_body 512
D (138017) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=14320
D (138027) transport_base: remain data in cache, need to read again
D (138037) HTTP_CLIENT: need_read=512, byte_to_read=512, rlen=496, ridx=512
D (138037) HTTP_CLIENT: http_on_body 496
D (138047) HTTP_CLIENT: http_on_message_complete, parser=3ffbe5d0
D (138047) HTTP_CLIENT: is_data_remain=0, is_chunked=0, content_length=14320
D (138067) esp_https_ota: Written image length 932336
D (138067) pre_encrypted_ota_example: Image bytes read: 932336
D (138067) HTTP_CLIENT: is_data_remain=0, is_chunked=0, content_length=14320
D (138077) esp_https_ota: Connection closed
E (138077) esp_encrypted_img: Invalid operation
E (138087) pre_encrypted_ota_example: esp_encrypted_img_decrypt_end() returned -1 = ESP_FAIL
E (138097) pre_encrypted_ota_example: ESP_HTTPS_OTA upgrade failed
```

ashelman-ec, 2022-12-19:
Tried downgrading to esp_encrypted_img `v2.0.1` (the version referenced in the example code), and to `v2.0.0`. Fails with the same error.

mahavirj, 2022-12-19:
@ashelman-ec 

I suspect this be related to the server setup and enabling "Enable partial HTTP download" may not help here.

Can you please revert to the default configuration and try uploading the image to different server, e.g. github.com? Here are instructions for that: https://github.com/espressif/esp-idf/tree/master/examples/system/ota#local-https-server

kitty7c6, 2023-10-20:
Hello! I write my own app: pre ecnrypted Ota by nimble. At first I tried Ota by nimble without encryption - it works. Than I add encryption, and also have error `esp_encrypted_img: Invalid operation`.
If I ignore this error (don't use esp_encrypted_img_decrypt_end()) - everything is allright and firmware updates well. But I have this error always, if I use esp_encrypted_img_decrypt_end().

For tests:
I  tried esp_encrypted_img_is_complete_data_received() before esp_encrypted_img_decrypt_end(), and got False
So then I I writed to log:
 handle->binary_file_len = 578096
 handle->binary_file_read = 578099
handle->cache_buf_len = 3

But data transfer is completed for this moment. What to do with this 3 bytes? And why thay are here

Than I used key from [examples/system/ota/pre_encrypted_ota/](https://github.com/espressif/esp-idf/tree/master/examples/system/ota/pre_encrypted_ota)
I also used this code as an example:
https://github.com/hmalpani/esp-idf/blob/feature/encrypted_image/examples/system/ota/pre_encrypted_binary_ota/main/main.c

hmalpani, 2023-10-27:
Hello @kitty7c6

I think in your case, you are sending more data to decrypt than the actual image size which is causing the error. Please check for this.
I tried the example in the link you shared and it worked fine.
BTW we also have a component which supports OTA over BLE: [ble_ota](https://components.espressif.com/components/espressif/ble_ota). You can check it out.

kitty7c6, 2023-10-31:
Hello @hmalpani ! I saw that component and tried it, and example compiled. But I need to choose config "CONFIG_EXAMPLE_USE_PRE_ENC_OTA", but in menuconfig there are only:
```
#
# Example Configuration
#
# CONFIG_EXAMPLE_OTA_SECURITY_VERSION_1 is not set
CONFIG_EXAMPLE_OTA_SECURITY_VERSION_2=y
CONFIG_EXAMPLE_OTA_SEC2_DEV_MODE=y
# CONFIG_EXAMPLE_OTA_SEC2_PROD_MODE is not set
CONFIG_EXAMPLE_OTA_WITHOUT_SECURITY=y
# end of Example Configuration
```

Where I can get PRE_ENC_OTA or what to choose in menuconfig so that there is a choice different types of OTA?

I use esp32s3 + vs code, espressif idf v1.6.5 and  esp_idf 5.1_dirty

hmalpani, 2023-10-31:
Hello @kitty7c6 

You have to enable Nimble for using pre_encrypted ota with this component. You can refer to this project configuration: https://github.com/espressif/esp-iot-solution/blob/master/examples/bluetooth/ble_ota/README.md#configure-the-project-2

Hope it helps.