espressif/idf-extra-components #120: led_strip example

franz-ms-muc, 2023-01-04:
as there is no Example for LED Strip, i made one: 

https://gitlab.com/meisterschulen-am-ostbahnhof-munchen/esp32-4diac-example-application/-/tree/LEDRainbow/TestLEDStrip

Enjoy !

suda-morris, 2023-01-05:
FYI, there's also an led strip use case in the IDF's blink example: https://github.com/espressif/esp-idf/blob/master/examples/get-started/blink/main/blink_example_main.c#L35

dremsol, 2023-01-19:
When calling following function for my RGBW strip in the led_strip example
````
led_strip_set_pixel_rgbw(led_strip, 0, 16, 16, 16, 16);
```
I'm running into following error
```
I (11358) example: Turning the LED ON!
E (11358) led_strip_rmt: led_strip_rmt_set_pixel_rgbw(48): wrong LED pixel format, expected 4 bytes per pixel
```
using following config
```
led_strip_config_t strip_config = {
    .strip_gpio_num = BLINK_GPIO,
    .max_leds = 270, // at least one LED on board
    .led_model = LED_MODEL_SK6812,
    };
```
What could i be missing here?

suda-morris, 2023-01-28:
Hi @dremsol looks like you missed setting the `led_pixel_format` in the `led_strip_config_t`?

dremsol, 2023-01-28:
Thanks! Solved

closed 2023-02-01 by suda-morris