espressif/idf-extra-components #129: Eigen .gitignore is (was) misconfigured

kutukvpavel, 2023-01-19:
Gitignore file for Eigen includes entry `*build*`. It leads to scripts/buildtests.in being ignored. In turn, this causes build failure if the component is cloned as a part (dependency) of a project repository, and Eigen dependency has to be removed and reinstalled using `idf.py` for every new local instance.

The issue has been fixed upstream, but there is no release yet. Is there any way to integrate this fix? I think a backport is required to do so, it's easy to create one, but where to host it?

igrr, 2023-01-25:
> In turn, this causes build failure if the component is cloned as a part (dependency) of a project repository, and Eigen dependency has to be removed and reinstalled using idf.py for every new local instance.

Could you please explain this part in a bit more detail? What steps did you perform to add Eigen to the project, and what is the build error that you observe?

kutukvpavel, 2023-01-25:
Well, initially I used `idf.py add-dependency` commands as suggested in the component registry [page](https://components.espressif.com/components/espressif/eigen), and observed only the issue described in my pull request #128. This issue forced me to move Eigen from (I can't remember exact name of the directory idf-component-manager uses, but it sounded something like this) auto-managed components to local components to make necessary edits. To accomplish this I simply moved Eigen folder into `components/` directory (as shown in [the docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#example-project)) and also, I believe, removed the component hash file.

After fixing #128 for my local copy of Eigen, I observed no build issues on the machine I initially created the project on, let's call it "the first machine" for short.

However, I use multiple machines for development of my project and, obviously, I use git to manage my sources. Therefore I initialized a git repo in my project directory (with `.gitignore` configured to ignore build directories only, so `components/` is indexed), pushed into a remote (sorry, private repo on a private server, so I can't just show it to you) and and then cloned the project repository to all my other computers. My understanding is, that cloned working copies should build with no issues, however, I faced `error configuring component Eigen` CMAKE error message for all cloned repositories.

Upon further inspection I found that initial error that lead to bailing out was `no such file or directory: scripts/buildtests.in`. I was surprised to find that `buildtests.in` exists on the "first machine", but is not indexed by git, and therefore is indeed missing in all cloned repos. `git-check-ignore` lead me to `.gitignore` file for Eigen and `*build*` line, then some searching lead me to [this](https://gitlab.com/libeigen/eigen/-/merge_requests/1149) merge request upstream, that got merged not long ago.

igrr, 2023-01-25:
Thanks for the explanation, I think now I understand what happened there.

Since #128 is now merged and the new component version is released, can you go back to using the version of the component installed by the component manager? I am assuming you will keep the `managed_components` directory in .gitignore, so the components downloaded by the component manager won't be checked into git.

kutukvpavel, 2023-01-26:
Frankly, I didn't even give component manager another thought ever since I switched to a local copy. Thanks for a reminder.

Indeed this seems to be the most sensible solution, until a new release of Eigen appears anyway. Closing.

closed 2023-01-26 by kutukvpavel