espressif/idf-extra-components #136: can't compile usb_host_ch34x_vcp

m4rk0m, 2023-02-06:
m1 MacBook 
```
idf.py build
Executing action: all (aliases: build)
Running cmake in directory /Users/marko/esp/usb_host_ch34x_vcp/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/Users/marko/.espressif/python_env/idf5.1_py3.9_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /Users/marko/esp/usb_host_ch34x_vcp"...
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is AppleClang 14.0.0.14000029
-- The CXX compiler identification is AppleClang 14.0.0.14000029
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:1 (idf_component_register):
  Unknown CMake command "idf_component_register".

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.25)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
See also "/Users/marko/esp/usb_host_ch34x_vcp/build/CMakeFiles/CMakeOutput.log".
See also "/Users/marko/esp/usb_host_ch34x_vcp/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1, output of the command is in the /Users/marko/esp/usb_host_ch34x_vcp/build/log/idf_py_stderr_output_88136 and /Users/marko/esp/usb_host_ch34x_vcp/build/log/idf_py_stdout_output_88136
```

tore-espressif, 2023-02-06:
Hi @m4rk0m ,
The `usb_host_ch34x_vcp` is not a buildable project, it is only a component that can be added to your project.
An example usage can be found here: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/cdc/cdc_acm_vcp

m4rk0m, 2023-02-07:
I got it thanks, it works for ch430, but when I connect Arduino Mega it throws an error even after I change the pid and vid. Isn't the Arduino also using cdc acm protocol? Arduino pid 0x0042 and vid 0x2341

m4rk0m, 2023-02-09:
Yea I tried but still same. I've looked it up and arduino mega uses ATMEL mega16u2 chip for serial communication not ftdi. 
This is the error I get: 

```
I (293) main_task: Calling app_main()
I (293) VCP example: Installing USB Host
I (333) VCP example: Installing CDC-ACM driver
I (333) VCP example: Opening any VCP device...
E (703) HUB: Bad transfer status: CHECK_SHORT_PROD_STR_DESC
E (703) HUB: Stage failed: CHECK_SHORT_PROD_STR_DESC
```

tore-espressif, 2023-02-09:
@m4rk0m We fixed the enumeration issues with `Stage failed: CHECK_SHORT_PROD_STR_DESC` internally, it should get public in upcoming days. I'll keep you posted

m4rk0m, 2023-02-09:
great, thanks

tore-espressif, 2023-02-22:
@m4rk0m The fix https://github.com/espressif/esp-idf/commit/7010349a9aa22785b7ddc1f80df982185bedd6e5 is in master, could you please try again?

m4rk0m, 2023-02-22:
still:
```
I (345946) VCP example: Opening any VCP device...
E (347146) HUB: Bad transfer status 4: CHECK_SHORT_PROD_STR_DESC
E (347146) HUB: Stage failed: CHECK_SHORT_PROD_STR_DESC
```

slightly different with the status 4...

also tried with another Arduino with ch340 and it sabotages connection every few outputs:

```
I (122672) VCP example: Opening any VCP device...

test 143
test 144
test 145
test 146
test 147
test 148
tesW (122972) CDC_ACM: Unsupported notification type 0xBD
I (122972) CDC_ACM: 46 bd ff ee 

test 576
test 577
test 578
tW (122982) CDC_ACM: Unsupported notification type 0x3F
I (122982) CDC_ACM: 46 3f ff ee 
```

tore-espressif, 2023-02-22:
It seems that you will need a specific driver for you USB device. At the moment, we don't have resources to implement it. We can leave this issue open and come back to it, if there are more users requesting this feature

closed 2023-11-21 by tore-espressif