espressif/idf-extra-components #139: MIDI stuff mostly disappeared compared to version that was integrated in idf 5.0

peergum, 2023-02-09:
Haven't gone through all the details, but the MIDI part doesn't compile anymore.

From a quick look, it's like all the MIDI-related parts that were in commit 871ecb8163 of the idf 5.0 have disappeared, inclusive some defines, config descriptors, etc...

I'll try to figure the missing bits, but that's quite a step back. Was that intentional?

tore-espressif, 2023-02-24:
Hello @peergum ,

I suppose you mean this commit https://github.com/espressif/esp-idf/commit/2dfaec31067e79d4ecec9e666ac8e48aa8b5f13e

> Was that intentional?

Yes, it was. The reasoning was that we don't want to provide 'wrappers' for flexible classes (such as HID or MIDI) if there is no tight coupling with the rest of ESP system.

In case of MIDI, you can directly use TinyUSB MIDI interface that allows you to configure the driver for far more complex usecases. The downside is that you must provide your own configuration descriptor.

What do you think? Is there any specific feature that you would like to see implemented (apart from a default implementation of config descriptor)?

peergum, 2023-02-24:
Hey Thomas,

Thanks for the clarification, and I understand your point, since TinyUSB is another project completely and it wouldn't make sense to duplicate the work. It felt a bit like a regression to me, but my limited knowledge and experience of both TinyUSB and MIDI are the reasons I asked. I reapplied the removed code and fixed stuff intuitively and saved the modified component in the components folder to override the managed one, and that did the trick.

I think some examples of proper (basic) MIDI implementation might be useful for people starting with this. I personally adapted some code from user [MidiBox](https://github.com/midibox/esp32-idf-uartmidi) and it kind of works, but issue is more related to MIDI and USB protocols than to the ESP in itself, so that may be out of the scope of this repo.

Thanks,
Phil.

tore-espressif, 2023-03-08:
@peergum thank you for the feedback, we will discuss the topic internally. I'll close the issue for now.
Feel free to reopen.
Cheers, Tomas

closed 2023-03-08 by tore-espressif