espressif/idf-extra-components #141: USB Hid String descriptors

fink-at-trmc-dk, 2023-02-10:
Hi,
I'm having trouble setting up a custom HID device using the TUSB library and hope that you can help :-)
The first part of the problem is that the windows host driver asks for more string descriptors than are defined. There might be something wrong with my descriptors or somewhere else - but I hope you are able to guide me in the right direction.

Sorry about the very long log post but it is quite visible what happens here. I have also attached the eclipse on windows based project.

MCU: ESP32-S3
ESP-IDF: ESP-IDF v5.1-dev-3296-g76433e4cb2

Please let me know if you need more input :-)
```
â†�[0;33m--- idf_monitor on \\.\COM9 115200 ---â†�[0m
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (17) boot: ESP-IDF v5.1-dev-3296-g76433e4cb2 2nd stage bootloader
I (17) boot: compile time Feb 10 2023 11:21:32
I (18) boot: chip revision: v0.1
I (22) boot.esp32s3: Boot SPI Speed : 80MHz
I (27) boot.esp32s3: SPI Mode       : DIO
I (31) boot.esp32s3: SPI Flash Size : 2MB
I (36) boot: Enabling RNG early entropy source...
I (41) boot: Partition Table:
I (45) boot: ## Label            Usage          Type ST Offset   Length
I (52) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (60) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (67) boot:  2 factory          factory app      00 00 00010000 00100000
I (75) boot: End of partition table
I (79) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0b620h ( 46624) map
I (96) esp_image: segment 1: paddr=0001b648 vaddr=3fc91800 size=026ach (  9900) load
I (98) esp_image: segment 2: paddr=0001dcfc vaddr=40374000 size=0231ch (  8988) load
I (106) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1bfach (114604) map
I (133) esp_image: segment 4: paddr=0003bfd4 vaddr=4037631c size=0b488h ( 46216) load
I (149) boot: Loaded app from partition at offset 0x10000
I (149) boot: Disabling RNG early entropy source...
I (161) cpu_start: Pro cpu up.
I (161) cpu_start: Starting app cpu, entry point is 0x40375260
0x40375260: call_start_cpu1 at C:/esp-idf/esp-idf-master/esp-idf/components/esp_system/port/cpu_start.c:152

I (0) cpu_start: App cpu up.
I (175) cpu_start: Pro cpu start user code
I (175) cpu_start: cpu freq: 160000000 Hz
I (175) cpu_start: Application information:
I (178) cpu_start: Project name:     tusb_hid
I (183) cpu_start: App version:      1
I (188) cpu_start: Compile time:     Feb 10 2023 11:21:09
I (194) cpu_start: ELF file SHA256:  931d17fde57b04c6...
I (200) cpu_start: ESP-IDF:          v5.1-dev-3296-g76433e4cb2
I (206) cpu_start: Min chip rev:     v0.0
I (211) cpu_start: Max chip rev:     v0.99 
I (216) cpu_start: Chip rev:         v0.1
I (221) heap_init: Initializing. RAM available for dynamic allocation:
I (228) heap_init: At 3FC94A58 len 00054CB8 (339 KiB): DRAM
I (234) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (241) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (247) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (254) spi_flash: detected ch
[tusb_hid.zip](https://github.com/espressif/idf-extra-components/files/10706464/tusb_hid.zip)
ip: generic
I (258) spi_flash: flash io: dio
W (262) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (275) sleep: Configure to isolate all GPIO pins in sleep state
I (282) sleep: Enable automatic switching of GPIO sleep configuration
I (289) app_start: Starting scheduler on CPU0
I (294) app_start: Starting scheduler on CPU1
I (294) main_task: Started on CPU0
I (304) main_task: Calling app_main()
I (304) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (314) tusb_desc: 
ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½
ï¿½ï¿½ï¿½  USB Device Descriptor Summary  ï¿½ï¿½ï¿½
ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½ï¿½
[... 308 lines cut ...]
  Get Descriptor String[2]
  Queue EP 80 with 42 bytes ...
USBD Xfer Complete on EP 80 with 42 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 02 03 09 04 0A 02
  Get Descriptor String[2]
  Queue EP 80 with 42 bytes ...
USBD Xfer Complete on EP 80 with 42 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes

USBD Setup Received 80 06 02 03 09 04 0A 02
  Get Descriptor String[2]
  Queue EP 80 with 42 bytes ...
USBD Xfer Complete on EP 80 with 42 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes
USBD Suspend : Remote Wakeup = 0
```

BR
Fink
[tusb_hid.zip](https://github.com/espressif/idf-extra-components/files/10706483/tusb_hid.zip)

fink-at-trmc-dk, 2023-02-12:
What I'm specifically is curious about here is why the host asks for string descriptor 238. Any ideas are welcome....
//Fink

roma-jam, 2023-02-12:
Hi @fink-at-trmc-dk, 

Descriptor with a special number 0xEE (238) is a Microsoft OS String Descriptor.

So it is absolutely OK that a device doesn't have such descriptor and handles it as you have provided in log (no descriptor -> STALL and proceed further work). 

If you would like to learn more about this, you can look at the information here: 
https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/microsoft-defined-usb-descriptors

If you have more questions feel free to ask them. 

Regards, 
RJ

fink-at-trmc-dk, 2023-02-12:
Ok Thanks for that. It did shed some light over the situation :-)
BR
Fink

closed 2023-02-12 by fink-at-trmc-dk