espressif/idf-extra-components #142: Custom HID Receives but does not send.

fink-at-trmc-dk, 2023-02-12:
Hi again,

Device: ESP32-S3
ESP-IDF: ESP-IDF v5.1-dev-3296-g76433e4cb2

Porting an application from STM32 to ESP32 I need to have the ESP32 behave as a Custom HID Device and talk to it via a C# program on windows.

IT enumerates fine and the host loads the driver and the PC application sends a message the ESP32 HID Device. This is visible in the log below. 

My problem is that it seems that the device does not send a response back. In the log below is seen that the response is queued as expected, but I cannot see the response in the PC Application at a breakpoint on incoming data (the original STM32 based device sends a response back and this is detected at this breakpoint in the C# application)

Any help on this matter is highly appreciated :-)

Below is the log showing this and the source for this application is attached.

```
I (17) boot: ESP-IDF v5.1-dev-3296-g76433e4cb2 2nd stage bootloader
I (17) boot: compile time Feb 12 2023 19:06:05
I (18) boot: chip revision: v0.1
I (22) boot.esp32s3: Boot SPI Speed : 80MHz
I (27) boot.esp32s3: SPI Mode       : DIO
I (31) boot.esp32s3: SPI Flash Size : 2MB
I (36) boot: Enabling RNG early entropy source...
I (41) boot: Partition Table:
I (45) boot: ## Label            Usage          Type ST Offset   Length
I (52) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (60) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (67) boot:  2 factory          factory app      00 00 00010000 00100000
I (75) boot: End of partition table
I (79) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0b250h ( 45648) map
I (95) esp_image: segment 1: paddr=0001b278 vaddr=3fc91800 size=026ach (  9900) load
I (98) esp_image: segment 2: paddr=0001d92c vaddr=40374000 size=026ech (  9964) load
I (106) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1bfb0h (114608) map
I (133) esp_image: segment 4: paddr=0003bfd8 vaddr=403766ec size=0b0b8h ( 45240) load
I (149) boot: Loaded app from partition at offset 0x10000
I (149) boot: Disabling RNG early entropy source...
I (160) cpu_start: Pro cpu up.
I (161) cpu_start: Starting app cpu, entry point is 0x40375260
0x40375260: call_start_cpu1 at C:/esp-idf/esp-idf-master/esp-idf/components/esp_system/port/cpu_start.c:152

I (0) cpu_start: App cpu up.
I (175) cpu_start: Pro cpu start user code
I (175) cpu_start: cpu freq: 160000000 Hz
I (175) cpu_start: Application information:
I (178) cpu_start: Project name:     tusb_hid
I (183) cpu_start: App version:      1
I (188) cpu_start: Compile time:     Feb 12 2023 19:05:56
I (194) cpu_start: ELF file SHA256:  cbec041d535be8f0...
I (200) cpu_start: ESP-IDF:          v5.1-dev-3296-g76433e4cb2
I (206) cpu_start: Min chip rev:     v0.0
I (211) cpu_start: Max chip rev:     v0.99 
I (216) cpu_start: Chip rev:         v0.1
I (220) heap_init: Initializing. RAM available for dynamic allocation:
I (228) heap_init: At 3FC94A58 len 00054CB8 (339 KiB): DRAM
I (234) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (240) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (247) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (254) spi_flash: detected chip: generic
I (258) spi_flash: flash io: dio
W (261) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (275) sleep: Configure to isolate all GPIO pins in sleep state
I (282) sleep: Enable automatic switching of GPIO sleep configuration
I (289) app_start: Starting scheduler on CPU0
I (294) app_start: Starting scheduler on CPU1
I (294) main_task: Started on CPU0
I (304) main_task: Calling app_main()
I (304) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (314) tusb_desc: 
???????????????????????????????????
?  USB Device Descriptor Summary  ?
[... 267 lines cut ...]

USBD Setup Received 80 06 02 03 09 04 FF 00
  Get Descriptor String[2]
  Queue EP 80 with 42 bytes ...
USBD Xfer Complete on EP 80 with 42 bytes
  Queue EP 00 with 0 bytes ...
USBD Xfer Complete on EP 00 with 0 bytes
USBD Xfer Complete on EP 01 with 64 bytes
  HID xfer callback
I (14864) Usb: Out:
01 0C 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 I (14874) Usb: tud_hid_set_report_cb (0 0 0)
  Queue EP 01 with 64 bytes ...
I (14884) Usb: In:
01 02 03 04 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   Queue EP 81 with 5 bytes ...
I (14914) Usb: OK
USBD Xfer Complete on EP 81 with 5 bytes
  HID xfer callback
USBD Suspend : Remote Wakeup = 0
USBD Suspend : Remote Wakeup = 0
```

BR
Fink

[tusb_hid.zip](https://github.com/espressif/idf-extra-components/files/10717030/tusb_hid.zip)

roma-jam, 2023-02-12:
Hi @fink-at-trmc-dk,

It seems that you have just queued the data but did not actually launched the sending process. 
But that is just a guess.

May I ask you to share the part of the code that you use to send the data through HID IN endpoint exactly? 

Thanks and best wishes, 
RJ

fink-at-trmc-dk, 2023-02-12:
Hi,
Yes, this was also my initial guess, but I can't see where the actual sending should be triggered. 
The application code (minimal project) is attached in the initial post.
BR
Fink

roma-jam, 2023-02-13:
Hi @fink-at-trmc-dk, 

Thanks for attaching the application code. 
From the first glance we have a line "USBD Xfer Complete on EP 81 with 5 bytes" in log which means that the transfer with EP 81 has been completed. 
To say more, it needs to be investigate deeply to see what is happening. 

Meanwhile, you can try to solve the problem on your own with the following steps:
- Compare the EP which is used in C# and STM32 project to receive HID report. It should be 81 EP for both examples. In HID class there is a possibility to use EP0 for such task. Just to be sure that the interface configuration is the same in both projects;
- You have sent 5 bytes with EP81 where max size is 64. Probably, on the host side the packet would be fully received when the data size is 64. Maybe there is a sense to try 64 bytes packet instead of 5;
-  There is a possibility to use Wireshark to see what is going on in the USB data flow and see the packets that you need. It also can be helpful to distinguish the side where the problem is. 

I will return to this issue ASAP. 
Feel free to ask questions. 

Regards, 
RJ

Rasmus-Fink, 2023-02-13:
Hi again,
Thanks for taking your time on this!
The descriptors are copied from the old project, so I assume the endpoints being used are the same. I do not know if that assumption is safe, but I will investigate. This could potentially be an easy fix :-)
The result was the same if I tried sending 64 bytes, so I do not think that this would be the problem.
I will do my homework on your inputs regarding EP81/EP0 an return back later :-)
//Fink

Rasmus-Fink, 2023-02-14:
Update:
After using [(USB analyzer)](https://freeusbanalyzer.com/) I can clearly see that the data are indeed returned to the host. They are, however, not picked up by the application even though they are the same type (Interrupt, EP81, size etc.) This is quite weird but I do not think I can blame the ESP32 for this one.
I hereby humbly close the issue while thanking you for your time and pointers :-)
BR
Fink

roma-jam, 2023-02-14:
Hi @Rasmus-Fink, 
Happy to help. 

If there will be more strange or questioning behavior feel free to open an issue.  
Good luck. 

RJ

closed 2023-02-14 by fink-at-trmc-dk