espressif/idf-extra-components #145: strip_gpio_num does not accept GPIO_NUM_NC

franz-ms-muc, 2023-02-20:
Hi, 

as GPIO_NUM_NC is -1, 
the Assignment ot 

`uint32_t strip_gpio_num; /*!< GPIO number that used by LED strip */`

fails. 

in some Scenarios where a Configurable IO is wanted, we might want to temporarily connect it to NC.

suda-morris, 2023-02-21:
> in some Scenarios where a Configurable IO is wanted, we might want to temporarily connect it to NC.

So you want to update the GPIO number dynamically? If so, you may want to delete the led_strip object and then install it with the new GPIO number.

franz-ms-muc, 2023-02-21:
No. Not dynamic. Store in NVS. Load at Boot.

suda-morris, 2023-02-21:
> No. Not dynamic. Store in NVS. Load at Boot.

Then can we install the led_stirp driver by checking whether the GPIO number stored in the NVS is >=0?

franz-ms-muc, 2023-02-21:
you are right. could.

closed 2023-04-12 by suda-morris