espressif/idf-extra-components #149: Convert the tests from test components to component-specific test apps

igrr, 2023-02-20:
Combining multiple test components into a single app doesn't scale well. For example, we already can't build the test_app for esp32s2, since the app doesn't fit into RAM. For this and other similar reasons, we have mostly migrated from test components to the component-specific test apps in IDF.

The task is to investigate how to do this in idf-extra-components, make the required changes to the tests and the CI workflow.

- Use https://github.com/espressif/idf-build-apps to create the build matrix and perform the builds.
- Only build and tests the components which have changed in the PR, and components which had changes to their dependencies.
- Build and test the examples inside components as well. (A test app is very similar to an example, it just doesn't get uploaded to the registry along with the component.)

lijunru-hub, 2023-10-12:
In CMake, I can't find the -U option. How can I achieve the functionality of WHOLE_ARCHIVE in IDF 4.4? Could you provide a CMake template for this?

igrr, 2023-10-27:
@lijunru-hub In v4.4 you can do it as done [here](https://github.com/espressif/esp-idf/blob/5f257494c5866e5f9965c54c907c2588f3496ad1/components/newlib/CMakeLists.txt#L48-L49) and [here](https://github.com/espressif/esp-idf/blob/5f257494c5866e5f9965c54c907c2588f3496ad1/components/newlib/assert.c#L94-L97). Note that you need one such hook symbol per object file.

igrr, 2024-09-18:
This is probably becoming more relevant, as the number of components in the repo has increased and we do a lot of unnecessary builds/tests when only one of the components gets updated.