espressif/idf-extra-components #164: `esp_tinyusb` console redirect should be gated behind VFS being enabled

dmezh, 2023-04-11:
### Answers checklist.

- [X] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [X] I am using target and esp-idf version as defined in component's idf_component.yml
- [X] I have searched the [issue tracker](https://github.com/espressif/idf-extra-components/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

usb

### ESP-IDF version.

v5.0.1

### Development Kit.

Custom Board (ESP32-S3FN8)

### Used Component version.

1.1.0

### More Information.

With VFS disabled in menuconfig, `esp_tusb_init_console` eventually fails to redirect std streams and returns an error at runtime. This function should just not be available if VFS is not available. Maybe an `#ifdef CONFIG_VFS_SUPPORT...` and a note in the documentation?

tore-espressif, 2023-05-03:
Hello @dmezh , you are right, I'm sorry for the troubles.

Since this issue can be fixed with correct menuconfig option, we will fix it in upcoming esp_tinyusb update ~end of June.

dmezh, 2023-05-03:
Awesome, thank you @tore-espressif!

closed 2023-05-24 by esp-saurabhbansal