espressif/idf-extra-components #169: Why does led_strip_set_pixel function of the led_strip component use uint32_t for color values that are expected to be within 0-255 range?

idea--list, 2023-04-23:
### Answers checklist.

- [X] I have read the documentation of the component in question and the issue is not addressed there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

In a project of mine i am about to predefine a bunch of colors like
`uint32_t led_green_color[3] = {0, 80, 0};`

and then just
```
memcpy(event_color, led_green_color, sizeof(event_color));
led_strip_clear(led_strip);
led_strip_set_pixel(led_strip, 0, event_color[0], event_color[1], event_color[2]);
```

This way i can change colors easily without having to reassign all the 3 color components 1 by 1.

I wonder why does `led_strip_set_pixel` use uint32_t even if the example tells that r,g,b components should have a value in the range of 0-255. That leads to wasting memory. Predefining 20 colors with uint32_t arrays needs 4*3*20 = 240 bytes instead of just 1*3*20 = 60 bytes that would be needed by using uint8_t for color values.

So my question is: why not just use uint8_t values for color values instead of uint32_t?

suda-morris, 2023-04-28:
@idea--list Thanks for raising this question. I can't 100% recall the reason ATM but one of the reasons I can think of is that RGB is not always 888 in the order, in some situations, we also can see R10G10B10.

If you want to save memory, how about adding a wrapper function? e.g.

```c
esp_err_t my_led_strip_set_pixel(led_strip_handle_t strip, uint32_t index, uint8_t red, uint8_t green, uint8_t blue)
{
    // TODO
   return led_strip_set_pixel(strip, index, red, green, blue);
}
```

idea--list, 2023-04-28:
Thanks for the tip, i am already doing that. Will report if i notice any regression. I was just wondering what might be the reasoning for uint32_t in this use case.

closed 2023-04-28 by idea--list