espressif/idf-extra-components #172: Cannot mount MSC on ESP32-S2

AlbertoLopSie, 2023-04-24:
### Answers checklist.

- [X] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [X] I am using target and esp-idf version as defined in component's idf_component.yml
- [X] I have searched the [issue tracker](https://github.com/espressif/idf-extra-components/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

usb

### ESP-IDF version.

5.0.1

### Development Kit.

ESP32-S2-Saola-1

### Used Component version.

1.2.0

### More Information.

Built [tusb\_msc](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_msc) on an [ESP32-S2-Saola-1](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-saola-1-v1.2.html) board not being able to be recognized as USB drive.

I've tried on Mac OSX 10.15.7 (Catalina) and on Raspbian 10. On both systems I get the serial monitoring port but the MSC is not recognized nor mounted.

Due to the 4MB board leaves less than 1MB to Flash SPI, I've slightly modified the 'size' command to report it in KB and also display sector count and size.

Running the modified [tusb\_msc](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_msc) example:
``` sh
SP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6108,len:0x1764
load:0x4004c000,len:0xabc
load:0x40050000,len:0x31b4
entry 0x4004c1c0
I (21) boot: ESP-IDF v5.0.1-dirty 2nd stage bootloader
I (21) boot: compile time 16:36:11
I (21) boot: chip revision: v0.0
I (24) boot.esp32s2: SPI Speed      : 80MHz
I (29) boot.esp32s2: SPI Mode       : DIO
I (34) boot.esp32s2: SPI Flash Size : 4MB
I (39) boot: Enabling RNG early entropy source...
I (44) boot: Partition Table:
I (48) boot: ## Label            Usage          Type ST Offset   Length
I (55) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (62) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (70) boot:  2 factory          factory app      00 00 00010000 00100000
I (77) boot:  3 storage          Unknown data     01 81 00110000 00100000
I (85) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=0c844h ( 51268) map
I (108) esp_image: segment 1: paddr=0001c86c vaddr=3ffbebf0 size=016f4h (  5876) load
I (109) esp_image: segment 2: paddr=0001df68 vaddr=40022000 size=020b0h (  8368) load
I (116) esp_image: segment 3: paddr=00020020 vaddr=40080020 size=2eb3ch (191292) map
I (161) esp_image: segment 4: paddr=0004eb64 vaddr=400240b0 size=0ab38h ( 43832) load
I (178) boot: Loaded app from partition at offset 0x10000
I (178) boot: Disabling RNG early entropy source...
I (190) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (190) cpu_start: Pro cpu up.
I (210) cpu_start: Pro cpu start user code
I (210) cpu_start: cpu freq: 160000000 Hz
I (210) cpu_start: Application information:
I (213) cpu_start: Project name:     tusb_msc
I (218) cpu_start: App version:      1
I (222) cpu_start: Compile time:     Apr 24 2023 16:36:10
I (228) cpu_start: ELF file SHA256:  e46f86cf7da1e424...
I (234) cpu_start: ESP-IDF:          v5.0.1-dirty
I (240) cpu_start: Min chip rev:     v0.0
I (244) cpu_start: Max chip rev:     v1.99 
I (249) cpu_start: Chip rev:         v0.0
I (254) heap_init: Initializing. RAM available for dynamic allocation:
I (261) heap_init: At 3FFC1130 len 0003AED0 (235 KiB): DRAM
I (267) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (274) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (281) spi_flash: detected chip: generic
I (285) spi_flash: flash io: dio
I (289) cpu_start: Starting scheduler on PRO CPU.
I (294) example_main: Initializing storage...
I (294) example_main: Initializing wear levelling
I (304) example_main: Mount storage...
I (304) example_main: 
ls command output:
README.MD
I (314) example_main: USB MSC initialization
I (314) tusb_desc: 
┌─────────────────────────────────┐
│  USB Device Descriptor Summary  │
├───────────────────┬─────────────┤
│bDeviceClass       │ 239         │
├───────────────────┼─────────────┤
│bDeviceSubClass    │ 2           │
├───────────────────┼─────────────┤
│bDeviceProtocol    │ 1           │
├───────────────────┼─────────────┤
│bMaxPacketSize0    │ 64          │
├───────────────────┼─────────────┤
│idVendor           │ 0x303a      │
├───────────────────┼─────────────┤
│idProduct          │ 0x4002      │
├───────────────────┼─────────────┤
│bcdDevice          │ 0x100       │
├───────────────────┼─────────────┤
│iManufacturer      │ 0x1         │
├───────────────────┼─────────────┤
│iProduct           │ 0x2         │
├───────────────────┼─────────────┤
│iSerialNumber      │ 0x3         │
├───────────────────┼─────────────┤
│bNumConfigurations │ 0x1         │
└───────────────────┴─────────────┘
I (484) TinyUSB: TinyUSB Driver installed
I (484) example_main: USB MSC initialization DONE

Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.
esp32s2> 
```

Running some example commands:

``` sh 
esp32s2> size
Storage Capacity 1000KB  (2000 sectors of 512 bytes each)
esp32s2> read
Mass Storage Devices are one of the most common USB devices. It use Mass Storage Class (MSC) that allow access to their internal data storage.
In this example, ESP chip will be recognised by host (PC) as Mass Storage Device.
Upon connection to USB host (PC), the example application will initialize the storage module and then the storage will be seen as removable device on PC.
esp32s2> status
storage exposed over USB: No
esp32s2> expose
I (164714) example_main: Unmount storage...
esp32s2> status
storage exposed over USB: Yes
esp32s2> 
```

However, no drive is detected on any OS. Raspbian output:

``` sh
$ dmesg
[607433.526387] usb 1-1.3: new full-speed USB device number 7 using dwc_otg
[607433.662537] usb 1-1.3: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[607433.662560] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[607433.662578] usb 1-1.3: Product: CP2102N USB to UART Bridge Controller
[607433.662594] usb 1-1.3: Manufacturer: Silicon Labs
[607433.662610] usb 1-1.3: SerialNumber: 3cfd8672cb1ced11be7412cebbdd3192
[607433.831902] usbcore: registered new interface driver usbserial_generic
[607433.831972] usbserial: USB Serial support registered for generic
[607433.838232] usbcore: registered new interface driver cp210x
[607433.838321] usbserial: USB Serial support registered for cp210x
[607433.838521] cp210x 1-1.3:1.0: cp210x converter detected
[607433.844473] usb 1-1.3: cp210x converter now attached to ttyUSB0

$ lsusb
Bus 001 Device 008: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP2102/CP2109 UART Bridge Controller [CP210x family]
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9512 Standard Microsystems Corp. SMC9512/9514 USB Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

$ lsblk
NAME         MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0      179:0    0  3.7G  0 disk
├─mmcblk0p1  179:1    0  256M  0 part /boot
└─mmcblk0p2  179:2    0  3.4G  0 part /
mmcblk0boot0 179:32   0    4M  1 disk
mmcblk0boot1 179:64   0    4M  1 disk
```

I'm aware of [this notice](https://github.com/espressif/esp-idf/issues/10837#issuecomment-1508568357) and verified my build is using the mentioned new driver.

Any hint??

igrr, 2023-04-24:
> [607433.844473] usb 1-1.3: cp210x converter now attached to ttyUSB0

Could you describe the way you are connecting the board to the computer? It looks as if you are plugging the USB cable into the USB port of the development board.

AlbertoLopSie, 2023-04-24:
Hi Ivan!

**I just realised that!!!  I can't believe my stupidity!!!**

I connected a USB breakout board to pins GPIO/20 on the saola board and voila!!!!   A brand new 1MB mass storage device appeared!!!

I was seeking to get that USB device on the same USB port I was connecting to flash and monitor. Looking at the board schematic I've found thase pins are not connected by default (a pair of NC 0 Ohm resistors give the option to achieve it, but I can't figure the side effects)

I was in the belief the CP2102 was able to produce 2 different devices when plugged to a host: the MSC and the serial port. Isn't that possible?

Thanks!!

igrr, 2023-04-25:
> I was in the belief the CP2102 was able to produce 2 different devices when plugged to a host: the MSC and the serial port. Isn't that possible?

I don't think it is, sorry. You can check the datasheet of CP2102 yourself: it is only a USB to serial converter chip, there is no MSC function.

Newer versions of ESP32-S2 dev boards ([ESP32-S2-DevKit-C](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-s2-devkitc-1.html)) actually come with two USB connectors so that you can use either the CP2102 USB-to-serial converter or the built-in USB_OTG peripheral of ESP32-S2 depending on what you need, without having to attach a USB breakout board.

AlbertoLopSie, 2023-04-25:
Thank you very much Ivan.

It is clear the mistakes I was making. I was able bild and run the example with no problems!

Have a nice day!

closed 2023-05-03 by tore-espressif