espressif/idf-extra-components #202: usb_host_msc: 1.1.0 can't use with psram open

lijunru-hub, 2023-07-11:
### Answers checklist.

- [X] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [X] I am using target and esp-idf version as defined in component's idf_component.yml
- [X] I have searched the [issue tracker](https://github.com/espressif/idf-extra-components/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

usb

### ESP-IDF version.

release/master, release/5.0

### Development Kit.

esp32-s3-n8r8

### Used Component version.

usb_host_msc: 1.1.0

### More Information.

Using version 1.1.0 of usb_host_msc and enabling PSRAM (OCT) simultaneously in ESP-IDF results in a failure to mount USB flash drives. This issue can be reproduced in both the release/master and release/v5.0 branches, using the MSC example from the ESP-IDF. However, this problem is not present in version 1.0.4.

```
         iManufacturer:  USB 
         iSerialNumber: 0501ed72cb11563ad6431daea2a5e4f 
E (10530) USB_MSC: msc_bulk_transfer_zcpy(599): 
E (10533) USB_MSC_SCSI: bot_execute_command(253): 
E (10538) diskio_usb: scsi_cmd_read10 failed (-1)
fresult 1
ESP_ERROR_CHECK failed: esp_err_t 0x1701 (ERROR) at 0x42009172
0x42009172: msc_init at /home/esp/project/usb_msc_example/image_display_2/build/../main/app_msc.c:193 (discriminator 1)

file: "../main/app_msc.c" line 193
func: msc_init
expression: msc_host_vfs_register(msc_device, "/usb", &mount_config, &vfs_handle)

abort() was called at PC 0x4037ae2f on core 0
0x4037ae2f: _esp_error_check_failed at /home/esp/esp-idf5/components/esp_system/esp_err.c:47
```

lijunru-hub, 2023-07-11:
[image]

igrr, 2023-07-18:
The issue is that `scsi_cmd_read10` and `scsi_cmd_write10` may be called with PSRAM buffers as arguments, however `msc_bulk_transfer_zcpy` now requires that the buffer is DMA capable:
https://github.com/espressif/idf-extra-components/blob/b38322876c01c96ee24447a7bccf9b8082706baf/usb/usb_host_msc/src/msc_host.c#L599
We could try to introduce `msc_bulk_transfer` function which would check if the buffer is DMA capable, and call `msc_bulk_transfer_zcpy` if so. Otherwise it would allocate an intermediate buffer in internal RAM and call `msc_bulk_transfer_zcpy` for that buffer.

lijunru-hub, 2023-07-20:
Expect an early resolution and whether it can be resolved on USB_Host_ The MSC component automatically inserts a USB drive and establishes a VFS file system with user-provided parameters. When the USB drive is unplugged, it automatically initializes the relevant drivers, making it more convenient to use.

tore-espressif, 2023-09-04:
Hi guys, I'm sorry for the slow follow-up. The flaky version 1.1.0 was yanked https://components.espressif.com/components/espressif/usb_host_msc/versions/1.1.0|

There are more bugs reported for this version, they will be fixed in next version

0727fqh, 2023-09-12:
[image]

Hello. I copy/fix/usb/MSC - string - descriptors] (HTTP: / / https://github.com/espressif/idf-extra-components/tree/fix/usb/msc-string-descripto rs) code, but this problem occurred when I was testing, not enough heap memory? I checked the memory with ESP.getFreeHeap(), ESP.getFreePsram() and nothing seems to be abnormal

tore-espressif, 2023-09-12:
@0727fqh thank you for the test! Do you run any other tasks in the application that could use the internal RAM? (It fails on internal RAM allocation, as PSRAM is not DMA capable)

You can read more about heap debugging here https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/heap_debug.html

Maldus512, 2023-11-13:
Unfortunately I'm still facing this issue with both  version 1.1.1 and 1.0.4. The MSC example for ESP-IDF v5.0 returns the following error if I try to enable octal PSRAM:

```
I (1188) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1188) gpio: GPIO[10]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1228) Main: Waiting for USB stick to be connected
I (1608) Main: MSC device connected
I (1608) Main: connection...
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.10
bDeviceClass 0x0
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 64
idVendor 0x54c
idProduct 0x6b1
bcdDevice 1.00
iManufacturer 1
iProduct 2
iSerialNumber 3
bNumConfigurations 1
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 32
bNumInterfaces 1
bConfigurationValue 1
iConfiguration 0
bmAttributes 0x80
bMaxPower 200mA
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 4
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 2
        bInterfaceClass 0x8
        bInterfaceSubClass 0x6
        bInterfaceProtocol 0x50
        iInterface 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x1    EP 1 OUT
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x82   EP 2 IN
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
Device info:
         Capacity: 30203 MB
         Sector size: 512
         Sector count: 61857791
         PID: 0x 6B1 
         VID: 0x 54C 
         iProduct: Storage Media 
         iManufacturer: Sony 
         iSerialNumber: 5C5211712215220336 
E (2008) USB_MSC: msc_bulk_transfer_zcpy(599): 
E (2008) USB_MSC_SCSI: bot_execute_command(253): 
E (2018) diskio_usb: scsi_cmd_read10 failed (-1)
ESP_ERROR_CHECK failed: esp_err_t 0x1701 (ERROR) at 0x42009841
0x42009841: app_main at /home/maldus/Projects/Elstore/leo/main/main.c:273 (discriminator 1)
```

For now I am using `heap_caps_malloc` directly from PSRAM only for certain allocations and it works.

tore-espressif, 2023-11-27:
@Maldus512 Could you please check your `dependencies.lock` file? It should be in your project root directory.

From the log it seems that you are still using version 1.1.0.

closed 2023-11-27 by tore-espressif

alonbl, 2023-12-24:
Hi,
In which version is this solved?
I am using the following and still face the issue.
```yaml
  espressif/usb_host_msc:
    component_hash: 8e870f035e212b33d23a7d051e1cd6f10986c1b34ea855df9ce0c04ffa91e1ac
    source:
      service_url: https://api.components.espressif.com/
      type: service
    version: 1.1.1~1
```
esp-idf: 341a8f2d65dc36b4e869feadc6cc0a21c96b0378
Regards,

alonbl, 2023-12-24:
Hello,

The `examples/peripherals/usb/host/msc` also fails in 341a8f2d65dc36b4e869feadc6cc0a21c96b0378.

```
I (319) main_task: Started on CPU0
I (339) main_task: Calling app_main()
I (339) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (339) example: Waiting for USB flash drive to be connected
I (769) example: MSC device connected
ESP_ERROR_CHECK failed: esp_err_t 0x1701 (ERROR) at 0x42009ae2
0x42009ae2: app_main at /home/alonbl/esp-idf/examples/peripherals/usb/host/msc/main/msc_example_main.c:279 (discriminator 1)

file: "/IDF/examples/peripherals/usb/host/msc/main/msc_example_main.c" line 279
func: app_main
expression: msc_host_vfs_register(msc_device, MNT_PATH, &mount_config, &vfs_handle)

abort() was called at PC 0x40379d37 on core 0
0x40379d37: _esp_error_check_failed at /home/alonbl/esp-idf/components/esp_system/esp_err.c:50

Backtrace: 0x403759c6:0x3fc99630 0x40379d41:0x3fc99650 0x40380855:0x3fc99670 0x40379d37:0x3fc996e0 0x42009ae2:0x3fc99710 0x4203121f:0x3fc99850 0x4037a7d5:0x3fc99880
0x403759c6: panic_abort at /home/alonbl/esp-idf/components/esp_system/panic.c:472

0x40379d41: esp_system_abort at /home/alonbl/esp-idf/components/esp_system/port/esp_system_chip.c:93

0x40380855: abort at /home/alonbl/esp-idf/components/newlib/abort.c:38

0x40379d37: _esp_error_check_failed at /home/alonbl/esp-idf/components/esp_system/esp_err.c:50

0x42009ae2: app_main at /home/alonbl/esp-idf/examples/peripherals/usb/host/msc/main/msc_example_main.c:279 (discriminator 1)

0x4203121f: main_task at /home/alonbl/esp-idf/components/freertos/app_startup.c:208

0x4037a7d5: vPortTaskWrapper at /home/alonbl/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
```

Should I open a new issue? as I do not think the example is using PSRAM and still fails with the same error.

Thanks,

alonbl, 2023-12-28:
Thanks @tore-espressif, it used to work in the past using FAT32 formatted by Windows, I will try again with different setup.

fozzzgate, 2024-06-12:
@alonbl Were you ever able to get this working with Windows FAT32 formatted USB drives? I am having this same problem and I can't figure out what is causing it. Windows formatted USB drives are not accepted (0x1701 - ESP_ERR_MSC_MOUNT_FAILED) and USB drives formatted by usb_host_msc can be read fine on ESP32, but are corrupt and not accepted on Windows.

I have downgraded to ESP IDF 5.0.4 / usb_host_msc 1.0.0 and it is still happening. Everything was working fine yesterday with Windows-formatted USB drives until I upgraded to ESP IDF 5.2.2 and now everything is broken and I can't even go back to a working version.

alonbl, 2024-06-13:
@fozzzgate: There is some incompatibility between how windows is creating/deleting the partitions and how the driver decode the device. At the end I found that erasing the device completely using `dd if=/dev/zero of=/dev/xx count=10` then recreating the FAT32 makes it usable again.

fozzzgate, 2024-06-13:
@alonbl After many hours of troubleshooting, I found the cause - USB drives that are GPT partitioned (for example, if they were previously used as boot drives created by Rufus) and then reformatted to FAT32 cause the error. The solution is to activate GPT compatibility in the fatfs component by editing ffconf.h and making these changes:

#define FF_LBA64		1
#define FF_FS_EXFAT	1

and changing these values in sdkconfig/menuconfig:

CONFIG_FATFS_LFN_HEAP=y (or CONFIG_FATFS_LFN_STACK=y)
CONFIG_FATFS_MAX_LFN=255

The error message is really uninformative: (ESP_ERR_MSC_MOUNT_FAILED [0x1701]). At the very least, the error enum for FR_NO_FILESYSTEM (in ff.h) and ESP_ERR_MSC_MOUNT_FAILED (in msc_host.h) should specify that those error codes may be returned if a GPT partitioned drive is attempting to be mounted and GPT support was not compiled into fatfs.

fozzzgate, 2024-06-21:
@tore-espressif In order to keep people from wasting time chasing the cause of a cryptic error message, I recommend adding the following comment to:

1. Line 288 of ff.h in the fatfs component (the comment associated with the FR_NO_FILESYSTEM error code)
2. Line 19 of msc_host.h in the fatfs component (the comment associated with the ESP_ERR_MSC_MOUNT_FAILED error code)

"This error may be caused by a GPT-formatted partition. GPT support can be enabled in fatfs."

Also, the timeout in line 83 of msc_host.c (WAIT_FOR_READY_TIMEOUT_MS) should be changed to 5000ms, to conform to the USB specification (it occasionally causes problems with one of my USB drives).