espressif/idf-extra-components #218: lib uvc - isochronous mode transfer doesnt seem correct, causes callback to client on each partial frame packet (IDFGH-10727)

sxrap1, 2023-07-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I can connect to USB UVC Camera, ans start streaming,
Frame Callback has packet lengths around the 1 to 2K bytes length, should be closer to 61K bytes
I can see the bytes coming in, in stream.c _uvc_stream_callback, however this code looks wrong to build up the full packet, hence the callback getting scrambled data.

leeebo, 2023-09-14:
@sxrap1 If you only need the UVC/UAC function in your USB Host project/product, try the [usb_stream](https://components.espressif.com/components/espressif/usb_stream) also. Which is based on a lower-level HCD driver, incompatible with other ESP-IDF USB host drivers, but easier to use and debug.

tore-espressif, 2024-03-06:
@sxrap1 yes, what you see is intended behavior now. Your application is responsible for creating a frame from the partial data now.

We are currently in process of refactoring the code so it provides better and cleaner API and usage for users

tore-espressif, 2024-07-02:
Unfortunately, I was not able to reproduce the problem. All cameras that I tried returned full frame in the callback.

Do you have any more debugging information that could help us?