espressif/idf-extra-components #234: After removing the esp32s3 usb cable from the computer, the fatfs was not remounted

hongquan-prog, 2023-09-12:
### Answers checklist.

- [X] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [X] I am using target and esp-idf version as defined in component's idf_component.yml
- [X] I have searched the [issue tracker](https://github.com/espressif/idf-extra-components/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

usb

### ESP-IDF version.

ESP-IDF v5.1.1-233-gf0437b945f

### Development Kit.

Custom

### Used Component version.

espressif/esp_tinyusb: 1.4.0 espressif/tinyusb：0.14.3

### More Information.

```c
// Invoked when device is unmounted
void tud_umount_cb(void)
{
    printf("tud_umount_cb: %p\n", s_storage_handle->base_path);

    if (tinyusb_msc_storage_mount(s_storage_handle->base_path) != ESP_OK) {
        ESP_LOGW(TAG, "tud_umount_cb() mount Fails");
    }
}

// Invoked when device is mounted (configured)
void tud_mount_cb(void)
{
    tinyusb_msc_storage_unmount();
    printf("tinyusb_msc_storage_unmount : %p\n", s_storage_handle->base_path);
}
```
I did this by adding a print filesystem mount path to the code, and when the USB cable was removed from the computer, the mount path became a null pointer and had to be remounted manually for it to work.the serial output is as follows：

```txt
I (5856) TinyUSB: TinyUSB Driver installed
I (5861) main: USB initialization DONE
I (5864) main_task: Returned from app_main()
tinyusb_msc_storage_unmount : 0x0
tinyusb_msc_storage_unmount : 0x0
tinyusb_msc_storage_unmount : 0x0
tinyusb_msc_storage_unmount : 0x0
tinyusb_msc_storage_unmount : 0x0
tinyusb_msc_storage_unmount : 0x0
tinyusb_msc_storage_unmount : 0x0
tinyusb_msc_storage_unmount : 0x0
tinyusb_msc_storage_unmount : 0x0
tinyusb_msc_storage_unmount : 0x0
tinyusb_msc_storage_unmount : 0x0
tinyusb_msc_storage_unmount : 0x0
tinyusb_msc_storage_unmount : 0x0
```
In addition to this problem, by printing I also found that tud_umount_cb is not called after I unplugged the usb, the event type triggered after unplugging the usb is DCD_EVENT_SUSPEND, not DCD_EVENT_UNPLUGGED.

leeebo, 2023-09-14:
@hongquan-prog For a self-powered USB device, the VBUS monitor must be enabled. please refer [the docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/usb_device.html?highlight=tinyusb#self-powered-device)

hongquan-prog, 2023-09-14:
> @hongquan-prog For a self-powered USB device, the VBUS monitor must be enabled. please refer [the docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/usb_device.html?highlight=tinyusb#self-powered-device)

Thank you very much for your help, I didn't consider making a standalone powered device at first, I just added a serial port, which will still power the esp32 when the USB is unplugged!

closed 2023-09-18 by hongquan-prog