espressif/idf-extra-components #243: pid_ctrl: support use IQmath to compute

lijunru-hub, 2023-09-21:
### Describe the solution you'd like.

`int bldc_pid_operation(bldc_pid_t* pid, int currentVale)
{
    _iq16 q16Up, q16Ui, q16Ud;

    currentVale = ABS(currentVale);

    pid->error = (pid->SetPoint - currentVale);

    q16Up = _IQ16mpy(_IQ16(pid->P), _IQ16(pid->error));
    q16Ui = _IQ16mpy(_IQ16(pid->I), _IQ16(pid->error));
    q16Ud = _IQ16mpy(_IQ16(pid->D), _IQ16(pid->error - pid->error_prev));

    pid->Up = _IQ16toF(q16Up);
    pid->Ui += _IQ16toF(q16Ui);
    pid->Ud = _IQ16toF(q16Ud);
    LIMIT_OUT(pid->Ui, pid->UiMax, pid->UiMin);

    float output = pid->Up + pid->Ui + pid->Ud;
    LIMIT_OUT(output, pid->OutMax, pid->OutMin);
    pid->error_prev = pid->error;
    return (int)output;
}`

### Additional context.

In chips like the esp32c3 that lack a floating-point computation unit, using floating-point calculations can incur significant system overhead

lijunru-hub, 2023-09-21:
After testing, it seems that modifying it to use IQmath calculations does not accelerate the PID computation time. Using IQmath on the ESP32C3 is actually slower than not using it.

It's worth mentioning that with the same code, the computation speed on ESP32S3 is 11 times faster than on ESP32C3. Is there a way to make the computation on ESP32C3 faster?

[image]

igrr, 2023-10-04:
On ESP32-C2, C6 and P4, there is an alternative floating point library ([rvfplib](https://github.com/pulp-platform/RVfplib)) in ROM. You can try enabling it using CONFIG_COMPILER_FLOAT_LIB_FROM_RVFPLIB option in menuconfig.

If enabling this option helps, we can open a feature request in IDF to support rvfplib also on chips which don't have it in the ROM (C3, H2).

who-has-seen-the-wind, 2023-10-11:
> On ESP32-C2, C6 and P4, there is an alternative floating point library ([rvfplib](https://github.com/pulp-platform/RVfplib)) in ROM. You can try enabling it using CONFIG_COMPILER_FLOAT_LIB_FROM_RVFPLIB option in menuconfig.
> 
> ..., we can open a feature request in IDF to support rvfplib also on chips which don't have it in the ROM (C3, H2).

Yes, iggr, please do open that feature request (to support RVfplib on the C3, etc) ! 
Floating point operations are fundamental, so this makes so much sense given that chips like the C3 don't have it in ROM.