espressif/idf-extra-components #249: Advertised dual port mode not actually implemented

DaStoned, 2023-10-06:
### Answers checklist.

- [X] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [X] I am using target and esp-idf version as defined in component's idf_component.yml
- [X] I have searched the [issue tracker](https://github.com/espressif/idf-extra-components/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

Other

### ESP-IDF version.

v5.1.1

### Development Kit.

ESP32S3-DevKitC

### Used Component version.

esp_modem_usb_dte 1.1.0~1

### More Information.

Quoting public documentation for esp_modem_usb_dte 1.1.0:

`Dual port modems

Some modems provide two equivalent AT ports. One of the ports can be used for AT commands, while the other one can be used for network data. This way, you don't have to switch between command and data modes of one terminal.

To use this feature, specify interface number of the second port in esp_modem_usb_term_config.`

This feature does not seem to be implemented. Sure, I can set the secondary inteface in `esp_modem_usb_term_config` and `create_usb_dte()` will happily give me an USB DTE. But when I try to activate dual mode, it returns an error.

```
    struct esp_modem_usb_term_config usb_config =
        ESP_MODEM_DEFAULT_USB_CONFIG_DUAL(GSM_USB_VID, GSM_USB_PID, GSM_USB_IDX_CMD, GSM_USB_IDX_DATA);
    esp_modem_dte_config_t dte_usb_config =
        ESP_MODEM_DTE_DEFAULT_USB_CONFIG(usb_config);

    _uart_dte = create_usb_dte(&dte_usb_config);
    _modem = create_SIM7070_dce(&_modem_config, _uart_dte, _esp_netif);
...

    if(!_modem->set_mode(modem_mode::DATA_MODE)) {
       tag_error("Fail activate DATA_MODE");
       return false;
    }
```
I cannot issue commands to the USB-connected modem (SIM7070) when it's in data mode. When I go and look at relevant code in the component "esp_modem", there is absolutely nothing that actually implements a dual mode feature. 

Clearly `DTE::set_mode(modem_mode m)` does absolutely nothing when told to activate DUAL_MODE:
https://github.com/espressif/esp-protocols/blob/ea14e15a6f113b1335f2a2511119c63b897f354d/components/esp_modem/src/esp_modem_dte.cpp#L199

And `DCE_Mode::set_unsafe()` explicitly ignores DUAL_MODE:
https://github.com/espressif/esp-protocols/blob/ea14e15a6f113b1335f2a2511119c63b897f354d/components/esp_modem/src/esp_modem_dce.cpp#L95

Why are you advertising a feature that's not implemented?

tore-espressif, 2023-10-12:
@DaStoned Thanks for the issue.
I had it working on my setup, but I must admit that I tested only with 2 modems. 

First of all, you must make sure that your modem supports USB dual-mode. The easiest way to do so, is by plugging the modem to any PC and check whether it creates two AT serial ports. Not all modems support this.

There is a regression in esp_modem 1.0.2 that breaks this function. Could you please try with esp_modem 1.0.1? In your main idf_component.yml only define `esp_modem: "1.0.1"`

Here is a link to the 1.0.1 source code for your convenience https://github.com/espressif/esp-protocols/tree/modem-v1.0.1/components/esp_modem

DaStoned, 2023-10-12:
@tore-espressif - thank you for the quick response. Dual mode is tested and supposed to work - that's good to hear. 

My modem is a SIMCOM SIM7070G which creates 6 virtual serial ports when plugged into a Linux PC. Their documentation (_SIM7070_SIM7080_SIM7090 Series_Linux_Application Note_V1.02.pdf_) says:

```
Interface number | Endpoint Type | Function
0                | USB serial    | Diagnostic Interface
1                | USB serial    | GPS NMEA Interface
2                | USB serial    | AT port Interface
3                | USB serial    | QFLOG Interface
4                | USB serial    | DAM Interface
5                | USB serial    | Modem port Interface
```
I plugged in indexes 2 and 5 into the macro `ESP_MODEM_DEFAULT_USB_CONFIG_DUAL()`, started a PPP dialup with `_modem->set_mode(modem_mode::DATA_MODE)` and expected to be able to issue AT commands during an ongoing PPP session. This does not work, as I see that the AT commands are spit right into the raw PPP stream. So my conclusion is that dual port mode is not active.

Perhaps I'm using some part of the API wrong in dual port mode. E.g. how am I supposed to start a PPP session in dual port?

DaStoned, 2023-10-12:
I'll add the modem & PPP bringup code for reference:
```
    if(ESP_OK != esp_event_handler_register(IP_EVENT, ESP_EVENT_ANY_ID, &on_ip_event, this)) {
        tag_error("Fail register IP event handler");
        return false;
    }
    if(ESP_OK != esp_event_handler_register(NETIF_PPP_STATUS, ESP_EVENT_ANY_ID, &on_ppp_changed, this)) {
        tag_error("Fail register PPP event handler");
        return false;
    }

    /* Configure the PPP netif */
    _modem_config = ESP_MODEM_DCE_DEFAULT_CONFIG(CONFIG_GSM_PPP_APN);
    _netif_ppp_config = ESP_NETIF_DEFAULT_PPP();
    _esp_netif = esp_netif_new(&_netif_ppp_config);
    if(!_esp_netif) {
        tag_error("Fail create PPP netif");
        return false;
    }
    hwBringUp();
    struct esp_modem_usb_term_config usb_config =
        ESP_MODEM_DEFAULT_USB_CONFIG(GSM_USB_VID, GSM_USB_PID, GSM_USB_IDX_CMD);
        //ESP_MODEM_DEFAULT_USB_CONFIG_DUAL(GSM_USB_VID, GSM_USB_PID, GSM_USB_IDX_CMD, GSM_USB_IDX_DATA);
    usb_config.timeout_ms = GSM_USB_APPEAR_TIMEOUT_MS;
    esp_modem_dte_config_t dte_usb_config =
        ESP_MODEM_DTE_DEFAULT_USB_CONFIG(usb_config);
    dte_usb_config.dte_buffer_size = 2048; // Must fit PPPoS MTU (1500 B?)

    tag_info("Waiting for USB device connection...");

    _uart_dte = create_usb_dte(&dte_usb_config);
    if (!_uart_dte) {
        tag_error("Fail create USB DTE");
        return false;
    }
    _uart_dte->set_error_cb(usb_terminal_error_handler);
    _modem = create_SIM7070_dce(&_modem_config, _uart_dte, _esp_netif);
    if (!_modem) {
        tag_error("Fail create SIM7070 modem");
        return false;
    }

    command_result cmd_err;
    std::string str_out;
    // Module may take some time to start from cold boot. Wait until responsive.
    uint32_t timeout = bootSecs() + GSM_STARTUP_TIMEOUT_S;
    do {
        cmd_err = _modem->get_module_name(str_out);
        if (command_result::OK != cmd_err) {
            tag_error("Fail get module name: %s (%u)", modem_err_to_name(cmd_err), static_cast<int>(cmd_err));
            vTaskDelay(pdMS_TO_TICKS(1000));
        }
    } while (command_result::OK != cmd_err && bootSecs() < timeout);
    if (command_result::OK != cmd_err) {
        tag_error("Fail module wakeup in %lu s", GSM_STARTUP_TIMEOUT_S);
        return false;
    }
    tag_info("Module name: %s", str_out.c_str());

    cmd_err = _modem->get_imei(str_out);
    if (command_result::OK == cmd_err) {
        tag_info("IMEI: %s", str_out.c_str());
    } else {
        // Odd, but we'll proceed.
        tag_error("Fail read IMEI: %s", modem_err_to_name(cmd_err));
    }

    cmd_err = _modem->get_imsi(str_out);
    if (command_result::OK == cmd_err) {
        tag_info("IMSI: %s", str_out.c_str());
    } else {
        // No SIM card. We'll proceed, though success is not likely.
        tag_error("Fail read IMSI: %s", modem_err_to_name(cmd_err));
    }

    int attached = 0;
    timeout = bootSecs() + GSM_ATTACH_TIMEOUT_S;
    do {
        vTaskDelay(pdMS_TO_TICKS(1000));
        cmd_err = _modem->get_network_attachment_state(attached);
        if (command_result::OK == cmd_err) {
            if(attached == 1){
                tag_info("Network attached.");
                break;
            }else{
                tag_info("Waiting for network...");
                _modem->set_network_attachment_state(true);
            }
        }else{
            tag_error("Fail get attachment state: %s", modem_err_to_name(cmd_err));
        }
    } while (bootSecs() < timeout);
    if (!attached) {
        tag_error("Fail attach to network in %lu s", GSM_ATTACH_TIMEOUT_S);
        return false;
    }

    cmd_err = _modem->get_operator_name(str_out);
    if (command_result::OK == cmd_err) {
        tag_info("Operator: %s", str_out.c_str());
    } else {
        tag_error("Fail read operator: %s", modem_err_to_name(cmd_err));
    }

    cmd_err = _modem->at("AT+CPSI?", str_out, 3000);
    if (command_result::OK == cmd_err) {
        tag_info("UE info: %s", str_out.c_str());
    } else {
        tag_error("Fail read UE info: %s", modem_err_to_name(cmd_err));
    }

    int rssi, ber;
    vTaskDelay(pdMS_TO_TICKS(1000)); // Signal measurements take time
    cmd_err = _modem->get_signal_quality(rssi, ber);
    if (command_result::OK == cmd_err) {
        tag_info("Signal quality: RSSI=%d BER=%d", rssi, ber);
    } else {
        tag_error("Fail read signal quality: %s", modem_err_to_name(cmd_err));
    }

    if(!_modem->set_mode(modem_mode::DATA_MODE)) {
       tag_error("Fail activate DATA_MODE");
       return false;
    }
```

tore-espressif, 2023-10-23:
> Interface number | Endpoint Type | Function
0                | USB serial    | Diagnostic Interface
1                | USB serial    | GPS NMEA Interface
2                | USB serial    | AT port Interface
3                | USB serial    | QFLOG Interface
4                | USB serial    | DAM Interface
5                | USB serial    | Modem port Interface

The dual port mode assumes that the two ports are interchangeable. It doesn't seem to be the case in your modem.

You can easily test it out. In your application, open the USB modem with single interface 2 and single interface 5. They should both work the same.

Alternatively, you can connect your modem to a PC and send AT commands to interface 5.

In case interface 5 can't accept AT commands, your modem does not support dual mode.

DaStoned, 2023-10-23:
That's strange. When I conduct an experiment with a SIM7070G devkit connected to my Linux PC, those two ports _are_ interchangeable. I can open either port 2 or 5 with minicom and issue commands `AT+CGDCONT=1,"IPV4V6","terminal.apn"` and `ATD*99#`. This successfully brings up a dial-up connection on either port. 

```
Port /dev/ttyUSB4, 13:05:52

Press CTRL-A Z for help on special keys

at
OK
AT+CGDCONT=1,"IPV4V6","terminal.apn"
OK
ATD*99#
CONNECT 150000000

OK
ath
OK
```

But when I try the experiment that you suggested with my board where a module of the same model is connected to ESP32S3 via USB, then port 5 doesn't receive any reply to any commands issued through it.

DaStoned, 2023-10-24:
I talked to the SimCom FAE and they confirmed - SIM7070G is a dual-port modem:

> You should use interface 2 for AT cmd and interface 5 for modem PPP protocol, they can work in parallel

My experiments on Linux confirm this. So I suspect there's something going on in the Espressif stack which doesn't allow me to use port 5.

tore-espressif, 2023-10-24:
Thanks for all the information, I'll try to get the modem and test for regressions

DaStoned, 2023-10-24:
I'm not very far from you (Estonia), so I can ship my evaluation board to you if there's nothing easily available.

diplfranzhoepfinger, 2023-12-13:
i did test the 
https://github.com/espressif/esp-protocols/tree/modem-v1.0.1/components/esp_modem/examples/pppos_client
and it worked for a A7672E-FASE

i opened another issue, 
how we could use more than only Dual Port, 
but also the Diag Port and the NMEA Port.

tore-espressif, 2023-12-13:
@diplfranzhoepfinger Thanks for the confirmation.

It must something specific to SIM7070G then

diplfranzhoepfinger, 2023-12-13:
@tore-espressif you sit in Brno ? 

@david-cermak has a A7672E-FASE from me. so you could ask him for a Demo.

tore-espressif, 2023-12-13:
> you sit in Brno ? 
@david-cermak has a A7672E-FASE from me. so you could ask him for a Demo.

Yep, will do

diplfranzhoepfinger, 2023-12-13:
> @diplfranzhoepfinger Thanks for the confirmation.
> 
> It must something specific to SIM7070G then

SIM7070G has at least a Problem: 

the GPS and the Network cannot work simultaneous. 

this Reason we went from SIM7070G to A7672E-FASE

Regards, 

Franz

DaStoned, 2023-12-13:
> > @diplfranzhoepfinger Thanks for the confirmation.
> > It must something specific to SIM7070G then
> 
> SIM7070G has at least a Problem:
> 
> the GPS and the Network cannot work simultaneous. 

Just to clarify, we're not using GPS in our project, just network connection.

tore-espressif, 2024-01-16:
@DaStoned I got SimTech SIM7070G and could do more testing.

This is devices that appear on windows 10:
[image]

Interface 2 corresponds to AT port, which is correct and works fine.
Interface 5 creates a "modem" device. We don't have support for this class in `esp_modem`. 

For now, you must use single interface 2 and switch between CMD and DATA modes.
At the moment, we don't plan to implement a driver for USB interfaces such as the one in SIM7070G

tore-espressif, 2024-01-16:
I'll rename this issue so we can collect similar request here and possibly reconsider this feature in the future.

DaStoned, 2024-01-16:
OK, a negative answer is still much better than no answer :) Thank you. 
May I suggest a list of not supported modems in the README?

closed 2024-01-16 by DaStoned

tore-espressif, 2024-01-16:
> May I suggest a list of not supported modems in the README?

With the significant differences among modems with similar names (eg. SIM7070 and SIM7670) I'd prefer a list of _supported_ devices. We will add it in next release