espressif/idf-extra-components #254: usb host msc fails to build with NEWLIB_NANO_FORMAT set to yes

greenaddress, 2023-10-18:
### Answers checklist.

- [X] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [X] I am using target and esp-idf version as defined in component's idf_component.yml
- [X] I have searched the [issue tracker](https://github.com/espressif/idf-extra-components/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

usb

### ESP-IDF version.

v5.1.1

### Development Kit.

esp32s3 tdisplay s3

### Used Component version.

1.1.1

### More Information.

```
undefined reference to `_vfwprintf_r'
```

tore-espressif, 2023-11-30:
Hi @greenaddress . The USB drivers (MSC, HID and UVC) use `wprintf` to handle USB string descriptors, which are UTF-16 encoded. We will fix in upcoming releases

tore-espressif, 2024-02-06:
@greenaddress The error was actually in the MS example itself. It was fixed internally. This issue will be automatically closed when the commit is synced to esp-idf github