espressif/idf-extra-components #274: Is the magic number in Encrypted image header static or dynamic hash?

dingiso, 2023-12-03:
### Answers checklist.

- [X] I have read the documentation of the component in question and the issue is not addressed there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Dear esp_encrypted_image maintainers:

There is a Image Magic field at the top of the encrypted image header. I am confused about whether this field is static or dynamic.

https://github.com/espressif/idf-extra-components/blob/440f7d0c815eb9627420d7e3c40f9782e791b564/esp_encrypted_img/tools/esp_enc_img_gen.py#L17-L18
The comment of above codes says that it is a hash number of the image. However, this magic number is hardcoded in the python script and the [README](https://github.com/espressif/idf-extra-components/blob/440f7d0c815eb9627420d7e3c40f9782e791b564/esp_encrypted_img/README.md#encrypt-the-image) says that this script can be directly used to generate the encrypted image. 

So is the magic number a dynamic hash or a static value ? 

Thanks,
Dingisoul

mahavirj, 2023-12-04:
@dingiso 

Magic value in the header is static. The comment only explains how the magic bytes were derived.

dingiso, 2023-12-05:
Thanks, mahavirj. I get it

closed 2023-12-05 by mahavirj