espressif/idf-extra-components #284: How to use all 4 Lines of a SIMCOM A6772-FASE

diplfranzhoepfinger, 2023-12-13:
### Is your feature request related to a problem?

.

### Describe the solution you'd like.

.

### Describe alternatives you've considered.

.

### Additional context.

.

tore-espressif, 2023-12-13:
Hi @diplfranzhoepfinger thanks for issue.

Here's some information that might help you getting started:
* The rest of the ports (apart from 5-Modem port for PPP dial up) are not strictly related to 'modem' so I assume that they won't be implemented in `esp_modem`. cc @david-cermak 
* The interface 5-Modem port for PPP dial up, is a different way  to open a network connection. Since not all modems offer this interface and we already have implementation with AT commands, we do not plan to implement this
* The rest of the ports should act as regular COM ports. You can open them using [usb_host_cdc_acm](https://components.espressif.com/components/espressif/usb_host_cdc_acm) (which is also used for the AT interfaces - 4 and 5 in this case)
* You can open the interface with this function
``` c
esp_err_t cdc_acm_host_open_vendor_specific(uint16_t vid, uint16_t pid, uint8_t interface_num, const cdc_acm_host_device_config_t *dev_config, cdc_acm_dev_hdl_t *cdc_hdl_ret);
```
relevant example is here https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/cdc/cdc_acm_host
* ICYMI, we also have neat component for parsing the NMEA messages https://components.espressif.com/components/igrr/libnmea

> https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/cdc/cdc_acm_vcp seems not to help, as it does tooo much Driver stuff and all this, we have already a lot inside the above Example.

This is not the first time that this example confused someone, I'll try to describe better what is this example for. Thanks for the feedback!

diplfranzhoepfinger, 2023-12-13:
@tore-espressif 

Thanks, we will have a Look.

tore-espressif, 2024-02-06:
@diplfranzhoepfinger Do you any any other inputs on this or can we close?

rinstrum-franz, 2024-06-27:
@tore-espressif i did try, but it chrashes. 

i need to deep into this again. 
Sorry for slow feedback so far. 

i will reopen this issue once i go back into it.

diplfranzhoepfinger, 2024-09-27:
FYI: 

my trys were here: 

[https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-pppos_client/tree/feature/NMEA](https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-pppos_client/tree/feature/NMEA)

and here: 

[https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-pppos_client/tree/feature/NMEA2](https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-pppos_client/tree/feature/NMEA2)

i will do a fresh restart now here: 

[https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-pppos_client-II](https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-pppos_client-II)

hold on

diplfranzhoepfinger, 2024-10-09:
so, use a modified ESP DEVKIT C, 
like this one: 
https://github.com/espressif/esp-idf/issues/14595

diplfranzhoepfinger, 2024-10-09:
using: 

[https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/usb_host_lib](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/usb_host_lib)

i get this: 

https://gist.github.com/diplfranzhoepfinger/015861debc6c5fad1cb4512a8e53728e

diplfranzhoepfinger, 2024-10-09:
now, using a modified Version of: 

[https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/cdc/cdc_acm_host](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/cdc/cdc_acm_host)

visible here: 

[https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-pppos_client-II/blob/feature/USB_third_port_try1/pppos_client/components/nmea_int/usb_cdc_example_main.c](https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-pppos_client-II/blob/feature/USB_third_port_try1/pppos_client/components/nmea_int/usb_cdc_example_main.c)

# Ports (interface\_num)

## Port 0

## Port 1

## Port 2

## Port 3

## Port 4

## Port 5

diplfranzhoepfinger, 2024-10-09:
next Step: 

connect the SIMCOM Kit to a Linux, and testing something: 

/dev/ttyUSB0

constantly sending stuff. ...

might be the Diagnostic port for log messages

/dev/ttyUSB1

/dev/ttyUSB2

unable to open

/dev/ttyUSB3

seeems to be a AT Port. 

/dev/ttyUSB4

diplfranzhoepfinger, 2024-10-09:
@tore-espressif 

now, i found out Port 3 is the NMEA Port, 

and if i enabled the Modem before (with my Linux) to output NMEA Sentences, 

this Code here: [https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-pppos_client-II/tree/nmea_001](https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-pppos_client-II/tree/nmea_001)

does read NMEA sentences and output it to the Console. 

NOW: 

if i uncomment this Line: 

[https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-pppos_client-II/blob/731bfb68a805f49bf78e7c7ec85d8b5b184723e2/pppos_client/main/main.c#L16](https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-pppos_client-II/blob/731bfb68a805f49bf78e7c7ec85d8b5b184723e2/pppos_client/main/main.c#L16)

i get an Error.

diplfranzhoepfinger, 2024-10-09:
@tore-espressif 

use this Code: 

https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-pppos_client-II/tree/nmea_003

and the Log here: 

https://gist.github.com/diplfranzhoepfinger/cdf3f865467d4585ded174e1dd5b5aee

tore-espressif, 2024-10-09:
I assume you are using IDF version >5.0 and latest usb_host_cdc_acm driver v2.0.4 https://components.espressif.com/components/espressif/usb_host_cdc_acm/versions/2.0.4

The CDC driver opens the CDC interfaces based on the bInterfaceNumber in Configuration descriptor, for your device this can be 0-5.

### Interface 0&1:
These 2 interface are bound together with Interface Association Descriptor (so they belong to the same USB Function) with USB Class = 0xE0: `USB_CLASS_WIRELESS_CONTROLLER`. So:
* Interface 0 is not opened, as it does not contain 2 Bulk endpoints
* Interface 1 is opened and you can try to use it, but you must know what format do the data endpoint have. We do not have and do not plan support for Wireless Controller class

### Interfaces 2-5
Seem to be working OK

### Interface 6
Does not exist in this device

> for me this is obvious, as the esp_modem component does call usb_host_install again.

For usecases when USB host is used for other purposes the communication with `esp_modem`, you can set this parameter to false 
https://github.com/espressif/esp-usb/blob/9be78b9d60325c42038ef8da8bde0e1c51e84690/host/class/cdc/esp_modem_usb_dte/include/esp_modem_usb_config.h#L23

This way you can install USB Host library in you application main and it can be shared with multiple threads