espressif/idf-extra-components #285: coap: To allow enable both COAP_MBEDTLS_PSK and COAP_MBEDTLS_PKI

AxelLin, 2023-12-15:
### Is your feature request related to a problem?

I notice the COAP_MBEDTLS_PSK and COAP_MBEDTLS_PKI cannot be enabled at the same time now.
I'm working on a coap client, however, using COAP_MBEDTLS_PSK or COAP_MBEDTLS_PKI
depends on the server being used.
Is it possible to enable both COAP_MBEDTLS_PSK and COAP_MBEDTLS_PKI?

hmalpani, 2023-12-18:
@mrdeep1 Can we support this feature?

mrdeep1, 2023-12-18:
Providing the client can determine at the start of opening a new session whether it wants to use PSK or PKI (I.e. doing separate connection to 2 different servers) there is no reason (other than some code changes) that both PSK and PKI modes can be supported.

It would be helpful to understand how the client is going to invoke the _psk2() or _pki() session setup call.

AxelLin, 2023-12-18:
Hi @mrdeep1 

Currently using PSK or PKI is decided  at **compile time**.
This is not ok because using PSK or PKI depends on the server being used.
It should allow using PSK or PKI at runtime.

mrdeep1, 2023-12-18:
Agreed it is one or the other at compile time at present.

However the runtime needs to know which of the 2 to methods to select when initiating the session to the remote server.  If you can let me know how you are proposing to do that (choose the method) then I can suggest what code are needed.

FYI, the standard libcoap build example coap-client decides which of PSK or PKI (or none) depending on the provided runtime options.

mrdeep1, 2023-12-19:
@AxelLin Have you worked out how you plan to tell the client at runtime as to whether it is to use PSK or PKI?

AxelLin, 2023-12-20:
@mrdeep1 

I have no strong preference about how to configure the library.
As long as we can config the library to use PSK or PKI, it's fine for us.

Honestly, I'm quite new in coap library.
I'm not sure how other coap libraries handling such configuration issue.

AxelLin, 2023-12-27:
@mrdeep1 

I think how to decide pki or psk can be decided by application.
For example, in the libcoap/examples/coap-client.c, it uses below logic
to decide pki or psk. (We can use similar logic)

```
// for DTLS/TLS case
if (root_ca_file || ca_file || cert_file) {
        /* Setup PKI session */
} else if (identity || key) {
        /* Setup PSK session */
} else {
        /* No PKI or PSK defined, as encrypted, use PKI */
}
```
The libcoap component just need to allow enable both pki and psk.

mrdeep1, 2023-12-27:
@AxelLin Sure the code can reflect this.  However, at run time, you need to be able to set or unset the appropriate variables to take the correct path.  That is where the real challenge is that needs to be solved for doing what you want to do.

AxelLin, 2023-12-27:
> @AxelLin Sure the code can reflect this. However, at run time, you need to be able to set or unset the appropriate variables to take the correct path. That is where the real challenge is that needs to be solved for doing what you want to do.

I think that is not a problem.

AxelLin, 2023-12-28:
@mrdeep1 

I have verified testing PKI and PSK seperately and it works.
As I said in previous comment, I actually don't need to runtime switching PKI or PSK without reboot.
What I need is to allow users to config the device send data to server with PKI or PSK.
It needs reboot if the user changed the configuration.
I think current limitation is due to the Kconfig used in "espressif/coap" component,
upstream libcoap can support both PKI and PSK enabled.

Now just waiting for the "espressif/coap" component to allow enable both PKI and PSK.

mrdeep1, 2023-12-28:
> The point is I can allow user to config using PSK or PKI, changing server
> settings require reboot.

Ok, I better understand your issue now.  I will look at getting this changed shortly.

mrdeep1, 2024-01-02:
@AxelLin Please checkout #289 which I believe provides what you require.

AxelLin, 2024-01-02:
> @AxelLin Please checkout #289 which I believe provides what you require.

Hi @mrdeep1 

It works, thank you.

mrdeep1, 2024-01-02:
Thanks for confirming.

mahavirj, 2024-01-15:
Fixed with https://github.com/espressif/idf-extra-components/commit/a1681cdf40baec06a097b48ea50de7704cb5c964

closed 2024-01-15 by mahavirj