espressif/idf-extra-components #297: Version 3 is not upgrading Version 1,

andy-danieal, 2024-01-20:
### Answers checklist.

- [X] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [X] I am using target and esp-idf version as defined in component's idf_component.yml
- [X] I have searched the [issue tracker](https://github.com/espressif/idf-extra-components/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

Other

### ESP-IDF version.

ESP-IDF v5.2-beta1-263-ge49823f10c

### Development Kit.

ESP32S2

### Used Component version.

1.1.0

### More Information.

Issue regrading :- [ESP_DELTA_OTA](https://github.com/espressif/idf-extra-components/tree/master/esp_delta_ota)

We have Delta OTA to generate Version Base Patch - **Version 1** and After generate **Version 2** and **Version 3**. But when i updgrade **Version 1**  to **Version 2** is working but **Version 1** to **Version 3** is not working.

```
E (40167) https_delta_ota_example: SHA256 of current firmware differs from than in patch header. Invalid patch for current firmware
E (40167) https_delta_ota_example: Patch Header verification failed

```

hmalpani, 2024-01-22:
Hello @andy-danieal 
From what I understand from your description, you have 3 binaries: **Version 1**, **Version 2** and **Version 3**. You must have created patches using:
1) Version 1 and Version 2
2) Version 2 and Version 3

From the logs, we can see that `SHA256 of current firmware differs from than in patch header. Invalid patch for current firmware`. It means that the patch you are trying to apply was not built using the current binary as base binary. If you want to update from Version 1 to Version 3, you must create a patch with **Version 1** as base binary and **Version 3** as new binary. I hope this helps you solve the issue you are facing. Please let me know if you still face the issue.

andy-danieal, 2024-01-23:
Thanks for your replay. 

We have 10 devices with Base **Version-1**. and we released OTA for all devices with **Version-2** but some devices not updating because internet issue. After some day, We updated new OTA for all devices with **Version-3** 

The problem is those devices that have not updated to **Version-2** are not able to update from version 1 to version 3.

hmalpani, 2024-01-23:
>> The problem is those devices that have not updated to Version-2 are not able to update from version 1 to version 3.

@andy-danieal, you will have to create the patch file with base binary as **Version-1** and the new binary as **Version-3**.
How have you created the patch file for the updates?

andy-danieal, 2024-01-23:
Base Version:- *Version-1*

```
python esp_delta_ota_patch_gen.py --chip esp32s2 --base_binary version-1.bin --new_binary version-3.bin --patch_file_name https_delta_ota_patch.bin
```

Issue on device which is Version2. that is not upgrading on Version3.
--------------------------------

But when generate new OTA from below command that time is Version1 not directly updagrade Version3.

```
python esp_delta_ota_patch_gen.py --chip esp32s2 --base_binary version-2.bin --new_binary version-3.bin --patch_file_name https_delta_ota_patch.bin
```

hmalpani, 2024-01-23:
@andy-danieal 
Base binary means the binary which is currently running on the device.

If the running firmware on the device is **Version-1** and you want to update it to **Version-3**, you must use the following command for creating the patch:
`python esp_delta_ota_patch_gen.py --chip esp32s2 --base_binary version-1.bin --new_binary version-3.bin --patch_file_name https_delta_ota_patch.bin`

Similarly, If the running firmware on the device is **Version-2** and you want to update it to **Version-3**, you must use the following command for creating the patch:
`python esp_delta_ota_patch_gen.py --chip esp32s2 --base_binary version-2.bin --new_binary version-3.bin --patch_file_name https_delta_ota_patch.bin`

andy-danieal, 2024-01-23:
As per ESP OTA same file work both Version2 and Version3. and we need one OTA file for different OTA version. Because many device has different version so we can't generate different files for all devices.

hmalpani, 2024-01-23:
@andy-danieal 
The patch file is created based on **base binary** and the **new binary**. If the base binary is different on different devices, then the patch will be different for those devices. You can't have a single patch file for OTA upgrade to a particular version if the base version on different devices is not same.

andy-danieal, 2024-01-24:
@hmalpani,

For example,  V1 (Base binary)

Device1 -> V1
Device2 -> V1 -> V2 (Device2 already upgrade V2)
Device3 -> V1 -> V2 (Device2 already upgrade V3)

In ESP-OTA, we have upload only one version V3. then all devies has been upgrade into V3. But In ESP-Delta-OTA that not working only those devices is update which is already in V2 version. that time V1 to direct V3 ota is not working.

hmalpani, 2024-01-25:
Hello @andy-danieal 
With **delta_ota**, you cannot have one only version hosted on the server if the base version running on different devices is different. This is because the patch is created using two binaries i.e. **base binary** and the **new binary**. If you want to update a device with **Version 1** firmware to **Version 3** firmware, you have two options:
1) Update the device firmware from **Version 1** to **Version 2** using the patch file created for upgrade to **Version 2**.
2) Update the device firmware from **Version 1** to **Version 3** by creating a new patch file using **Version 1** as base binary and **Version 3** as new binary

Although if all the devices have same version of the firmware running, then you will only need to host one patch file which can be used for the upgrade by all the devices.

hmalpani, 2024-02-08:
I hope the above explanation helped you with your query. Closing this issue. Please feel free to reopen.