espressif/idf-extra-components #298: Onewire search fails for devices with different bit 0

rpaulbeam, 2024-01-20:
### Answers checklist.

- [X] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [X] I am using target and esp-idf version as defined in component's idf_component.yml
- [X] I have searched the [issue tracker](https://github.com/espressif/idf-extra-components/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

onewire_bus

### ESP-IDF version.

5.1.2

### Development Kit.

ESP32-S3-DevKitC

### Used Component version.

1.0.1

### More Information.

onewire_device_iter_get_next() fails if the discrepancy occurs on bit 0.   For multiple devices of the same family, this does not happen, but in my case I have a DS2408 and a MAX31820(same as ds18B20).   At the beginning of the search (line69), last_zero is set to 0, but this is also used to test for the last device detected (line 111).   So, it there is a real discrepancy at bit 0, the search takes the '1' direction, and then terminates thinking it has discovered all devices.   The App Note referenced sets last_zero = 0, but uses bit_id = 1 as the start condition -- this implementation uses rom_bit_index which starts at 0.

This project is also for the esp32 and uses a bit_id that starts at 1:  [https://github.com/SuperHouse/esp-open-rtos/blob/master/extras/onewire/onewire.c](url)  I have not verified it works, but I am going to implement something similar.  It is probably as simple as starting rom_bit_index at 1, and then dealing with all the -1s necessary to fix rom_bit_mask and rom_byte_index etc.

rpaulbeam, 2024-01-20:
I made the following changes and the search algorithm detects both of my devices now.   There may be a more elegant solution.

    for (uint16_t rom_bit_index = 1; rom_bit_index < ((sizeof(onewire_device_address_t) * 8) +1); rom_bit_index ++) {
        uint8_t rom_byte_index = (rom_bit_index-1) / 8;
        uint8_t rom_bit_mask = 1 << ((rom_bit_index-1) % 8); // calculate byte index and bit mask in advance for convenience

ramiws, 2024-01-27:
it also fails for one device of DS18B20 :

https://www.lcsc.com/product-detail/Temperature-Sensors_UMW-Youtai-Semiconductor-Co-Ltd-DS18B20_C376006.html

The same sensor works in the arduino library for S3 but not the IDF library. Is there any quick hack that I can do?

rpaulbeam, 2024-01-27:
What I did is here: https://github.com/espressif/idf-extra-components/issues/298

https://www.lcsc.com/product-detail/Temperature-Sensors_UMW-Youtai-Semiconductor-Co-Ltd-DS18B20_C376006.html

closed 2026-09-15 by suda-morris