espressif/idf-extra-components #305: One_wire with ds18b20  rmt_isr_handle_rx_done crash

ramiws, 2024-02-10:
### Answers checklist.

- [X] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [X] I am using target and esp-idf version as defined in component's idf_component.yml
- [X] I have searched the [issue tracker](https://github.com/espressif/idf-extra-components/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

onewire_bus

### ESP-IDF version.

5.0 stable 

### Development Kit.

Esp32 s3 devkit

### Used Component version.

1.0.1

### More Information.

```
assert failed: rmt_isr_handle_rx_done rmt_rx.c:503 (offset > rx_chan->mem_off)

Backtrace: 0x4037606e:0x3fc9b600 0x4037ea0d:0x3fc9b620 0x40385bbd:0x3fc9b640 0x4037819d:0x3fc9b760 0x4037827a:0x3fc9b790 0x40376669:0x3fc9b7b0 0x40377431:0x3fc9b7d0 0x4037c9b7:0x3fcf49e0 0x4200439e:0x3fcf4a00 0x403800f9:0x3fcf4a20 0x40382509:0x3fcf4a40
0x4037606e: panic_abort at /Users/rami/Documents/esp-idf/components/esp_system/panic.c:412

0x4037ea0d: esp_system_abort at /Users/rami/Documents/esp-idf/components/esp_system/esp_system.c:135

0x40385bbd: __assert_func at /Users/rami/Documents/esp-idf/components/newlib/assert.c:78

0x4037819d: rmt_isr_handle_rx_done at /Users/rami/Documents/esp-idf/components/driver/rmt/rmt_rx.c:503 (discriminator 1)

0x4037827a: rmt_rx_default_isr at /Users/rami/Documents/esp-idf/components/driver/rmt/rmt_rx.c:598

0x40376669: shared_intr_isr at /Users/rami/Documents/esp-idf/components/esp_hw_support/intr_alloc.c:436

0x40377431: _xt_lowint1 at /Users/rami/Documents/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1117

0x4037c9b7: xt_utils_wait_for_intr at /Users/rami/Documents/esp-idf/components/xtensa/include/xt_utils.h:81
 (inlined by) esp_cpu_wait_for_intr at /Users/rami/Documents/esp-idf/components/esp_hw_support/cpu.c:101

0x4200439e: esp_vApplicationIdleHook at /Users/rami/Documents/esp-idf/components/esp_system/freertos_hooks.c:59

0x403800f9: prvIdleTask at /Users/rami/Documents/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4157 (discriminator 1)

0x40382509: vPortTaskWrapper at /Users/rami/Documents/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151

ELF file SHA256: 92d233ed9c75bd71
```

I get this using the example code. There is no example project to test. I am using espressif/ds18b20: "^0.1.1"

The sensor is found "Found a DS18B20[0], address: B20623945BA15A28"
The crash does not happen if I remove the call to ds18b20_get_temperature.

Any ideas what is this about?

ramiws, 2024-02-10:
@suda-morris I do not have the RMT ISR config, i tried to toggle with no effect

suda-morris, 2024-02-18:
@ramiws can you upgrade your IDF version to the latest release/v5.0? This commit should have fixed the issue you met. https://github.com/espressif/esp-idf/commit/641221a10a356ac49236f3b4a41a82871551aa94

closed 2024-03-11 by suda-morris