espressif/idf-extra-components #311: Speed up tinyusb CDC serial transfer rate on ESP32-S3

Superberti, 2024-03-04:
### Answers checklist.

- [X] I have read the documentation of the component in question and the issue is not addressed there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I'm using an ESP32-S3 on ESP-IDF on the release/5.2 branch.
I made some tests with the tinyusb CDC serial profile but I was not able to get read transfer speeds on the native port (pins 19/20) over 50 KiB/s. The theoretical maximum rate should be something about 1 MiB/s in this mode.
Regardless what I'm doing (larger buffers and FIFOs, using tinyusb with or without tusb_config-h), it had no influence on the transfer speed. So it seems that the speed is limited in some other way. The USB port is connected to a C# test program which sends out the data as fast as it can, so this should not be a problem at all.
Any ideas how to speedup the CDC read speed?

Thanks and bye,
Oliver

tore-espressif, 2024-03-25:
Hello @Superberti Thank you much for the report.

We can reproduce the issue and see similar performance.
The bottleneck is in the DCD layer of TinyUSB, this will requires significant refactoring. We plan to work this in Q2/2024

Superberti, 2024-03-26:
Thank you for clarification on this subject!