espressif/idf-extra-components #315: Component "spi_nand_flash" with another spi slave device.

Coco1780, 2024-03-14:
### Answers checklist.

- [X] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [X] I am using target and esp-idf version as defined in component's idf_component.yml
- [X] I have searched the [issue tracker](https://github.com/espressif/idf-extra-components/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

Other

### ESP-IDF version.

5.0.4

### Development Kit.

Custom board

### Used Component version.

rathisonika/spi_nand_flash 0.0.7 and espressif/spi_nand_flash 0.1.0

### More Information.

First of all thank you for this very useful extension for me.
I use a W25N01GVZEIG flash which works very well on its own. 
But I also use a w5500 ethernet spi module on the same spi bus. The two together don't work. After some research I think I found the problem. In the "spi_nand_execute_transaction" function I replaced "spi_device_transmit" by "spi_device_polling_transmit" and everything works now.

BTW : I can't compile espressif/spi_nand_flash 0.1.0 I always get "dhara/map.h" not found

adokitkat, 2024-03-14:
@RathiSonika

RathiSonika, 2024-03-14:
Hello @Coco1780 , Thank you for bringing up the issue. I'll investigate the `spi_nand_execute_transaction` function.

Regarding the compilation problem, the spi_nand_flash component relies on the dhara library, which is included as a separate component in the component repository. If you're compiling the example within the idf_extra_components repository, please initialize the submodule using the command `git submodule update --init dhara/dhara`, and then try again. Alternatively, you can remove the override path from the example's idf_component.yml file and attempt to compile it again. This should resolve the issue.