espressif/idf-extra-components #323: SPI NAND Flash Driver  with AS5F38G04SND-08LIN memory

Jayegit, 2024-04-30:
### Answers checklist.

- [X] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [X] I am using target and esp-idf version as defined in component's idf_component.yml
- [X] I have searched the [issue tracker](https://github.com/espressif/idf-extra-components/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

Other

### ESP-IDF version.

v5.1.2

### Development Kit.

ESP32-S3

### Used Component version.

0.1.0

### More Information.

I tried Alliance  AS5F38G04SND-08LIN  Flash memory  with using spi_nand_flash (SPI NAND Flash Driver) Component Driver but its did not work. How can I solved tins issue. Please help me...

This is serial monitor out put of after flash the program:
E (371) nand: spi_nand_flash_init_device(191): Failed to detect nand chip
ESP_ERROR_CHECK failed: esp_err_t 0x108 (ESP_ERR_INVALID_RESPONSE) at 0x42008911
0x42008911: example_init_nand_flash at C:/Espressif/workspace2/nand_flash/main/spi_nand_flash_example_main.c:145 (discriminator 1)

file: "./main/spi_nand_flash_example_main.c" line 145
func: example_init_nand_flash
expression: spi_nand_flash_init_device(&nand_flash_config, &nand_flash_device_handle)

abort() was called at PC 0x4037a527 on core 0
0x4037a527: _esp_error_check_failed at C:/Espressif/frameworks/esp-idf-v5.1.2/components/esp_system/esp_err.c:50

Jayegit, 2024-05-02:
Do I have to edit something of source code to use  Alliance AS5F38G04SND-08LIN (1GB) SPI Nand Flash memory?

igrr, 2024-05-20:
@Jayegit Sorry for the late reply. In order to add support for this flash memory, you need to do changes similar to this other PR: https://github.com/espressif/idf-extra-components/pull/327. This involves adding a device-specific init function and call it based on the ID of the chip. We don't have this specific flash chip, but if you do the necessary changes and submit a PR, we can merge it.

Jayegit, 2024-05-28:
Thank you very much for your reply, 
I want to working 8 Gbit (1GB) Flash with ESP32-S3 in my project. please let me know what kind of 1GB Flash ROM  (maker/device) can use with ESP32-S3 with this IDF component. Thank you...

closed 2025-05-12 by RathiSonika