espressif/idf-extra-components #331: Request for Official TAR Pack/Unpack Library Support in ESP-IDF (IDFGH-12851)

hayschan, 2024-05-20:
### Is your feature request related to a problem?

I'm always frustrated when trying to handle compressed archives containing multiple files and directories on the ESP32, as the ESP-IDF currently supports `zlib`, `miniz`, `xy_compress` but does not have built-in support for handling tar archives. This limitation complicates the process of downloading and managing multiple files in a compressed format, which is a common requirement for many applications, such as firmware updates or content delivery.

### Describe the solution you'd like.

I would like an official ESP-IDF library for tar packing and unpacking. This library should be capable of creating and extracting tar archives, supporting both standard tar files and compressed tar.gz files using zlib for decompression. This addition would greatly simplify the process of managing multiple files and directories in a compressed format on the ESP32, enhancing the overall functionality of the ESP-IDF.

### Describe alternatives you've considered.

I have considered using third-party libraries like libtar and writing a custom tar unpacker, but these solutions require significant modifications and integration effort to work with ESP-IDF. Additionally, they may not be as optimized or well-supported as an official library would be. 

### Additional context.

An official tar pack/unpack library would be particularly useful for applications involving over-the-air updates, data logging, and content delivery, where managing multiple files in a compressed format is essential. This feature would enable developers to streamline their workflows and improve the efficiency of their ESP32-based projects.

Holofil, 2025-01-03:
My upvote for this feature support as well.

mrunalg, 2025-01-03:
My upvote for this feature as well.

tobozo, 2025-01-05:
not an esp-idf component, but very practical https://github.com/dsoprea/TinyUntar