espressif/idf-extra-components #340: [led_strip]  After initializing the WS2812 with c6, calling led_strip_clear will cause the LEDs to turn green.

lijunru-hub, 2024-06-13:
### Answers checklist.

- [X] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [X] I am using target and esp-idf version as defined in component's idf_component.yml
- [X] I have searched the [issue tracker](https://github.com/espressif/idf-extra-components/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

led_strip

### ESP-IDF version.

IDF release/v5.2

### Development Kit.

ESP32-C6-Devkit-1 v1.2

### Used Component version.

2.4.3

### More Information.

In the example **led_strip_spi_ws2812**, after powering on and initializing the led_strip, immediately calling **led_strip_clear** will cause the LEDs to turn green.

```c
led_strip_handle_t configure_led(void)
{
    // LED strip general initialization, according to your led board design
    led_strip_config_t strip_config = {
        .strip_gpio_num = LED_STRIP_BLINK_GPIO,   // The GPIO that connected to the LED strip's data line
        .max_leds = LED_STRIP_LED_NUMBERS,        // The number of LEDs in the strip,
        .led_pixel_format = LED_PIXEL_FORMAT_GRB, // Pixel format of your LED strip
        .led_model = LED_MODEL_WS2812,            // LED strip model
        .flags.invert_out = false,                // whether to invert the output signal
    };

    // LED strip backend configuration: SPI
    led_strip_spi_config_t spi_config = {
        .clk_src = SPI_CLK_SRC_DEFAULT, // different clock source can lead to different power consumption
        .flags.with_dma = true,         // Using DMA can improve performance and help drive more LEDs
        .spi_bus = SPI2_HOST,           // SPI bus ID
    };

    // LED Strip object handle
    led_strip_handle_t led_strip;
    ESP_ERROR_CHECK(led_strip_new_spi_device(&strip_config, &spi_config, &led_strip));
    ESP_ERROR_CHECK(led_strip_clear(led_strip));
    ESP_LOGI(TAG, "Created LED strip object with SPI backend");

    return led_strip;
}
```

[image]

lijunru-hub, 2024-06-13:
The key point is that led_strip_clear should not be called immediately after initialization. Inserting a print statement in between prevents the issue. However, this problem does not occur on the S3.

lijunru-hub, 2024-06-13:
code
```c
/*
 * SPDX-FileCopyrightText: 2023 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Unlicense OR CC0-1.0
 */
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "led_strip.h"
#include "esp_log.h"
#include "esp_err.h"

// GPIO assignment
#define LED_STRIP_BLINK_GPIO  8
// Numbers of the LED in the strip
#define LED_STRIP_LED_NUMBERS 1

static const char *TAG = "example";

led_strip_handle_t configure_led(void)
{
    // LED strip general initialization, according to your led board design
    led_strip_config_t strip_config = {
        .strip_gpio_num = LED_STRIP_BLINK_GPIO,   // The GPIO that connected to the LED strip's data line
        .max_leds = LED_STRIP_LED_NUMBERS,        // The number of LEDs in the strip,
        .led_pixel_format = LED_PIXEL_FORMAT_GRB, // Pixel format of your LED strip
        .led_model = LED_MODEL_WS2812,            // LED strip model
        .flags.invert_out = false,                // whether to invert the output signal
    };

    // LED strip backend configuration: SPI
    led_strip_spi_config_t spi_config = {
        .clk_src = SPI_CLK_SRC_DEFAULT, // different clock source can lead to different power consumption
        .flags.with_dma = true,         // Using DMA can improve performance and help drive more LEDs
        .spi_bus = SPI2_HOST,           // SPI bus ID
    };

    // LED Strip object handle
    led_strip_handle_t led_strip;
    ESP_ERROR_CHECK(led_strip_new_spi_device(&strip_config, &spi_config, &led_strip));
    ESP_ERROR_CHECK(led_strip_clear(led_strip));
    ESP_LOGI(TAG, "Created LED strip object with SPI backend");

    return led_strip;
}

void app_main(void)
{
    configure_led();

    ESP_LOGI(TAG, "Start blinking LED strip");
}

````

suda-morris, 2024-06-18:
Thanks for the reproduce code. It's fixed in led_strip >= 2.5.4

closed 2024-06-18 by suda-morris