espressif/idf-extra-components #357: SimpleIni (PACMAN-945)

diplfranzhoepfinger, 2024-07-29:
### Related area

Component https://github.com/brofield/simpleini

### Is your feature request related to a problem?

https://github.com/brofield/simpleini

### Describe the solution you'd like

would like this as a ESP Component

### Describe alternatives you've considered

.

### Additional context

.

### I have checked existing list of Feature requests and I have provided enough information.

- [X] I confirm I have checked existing list of Feature requests and I have provided enough information.

igrr, 2024-07-29:
There is one ini-related component contributed by @leebo already: https://components.espressif.com/components/leeebo/esp-inih.

(Note, however, the version is 0.0.2, and it hasn't been updated with the latest [upstream releases](https://github.com/benhoyt/inih), so I don't know how future-proof it is.)

diplfranzhoepfinger, 2024-07-29:
> There is one ini-related component contributed by @leeebo already: https://components.espressif.com/components/leeebo/esp-inih.
> 

totally true, i use it a lot. 
cool lib. 

but it can ONLY read *.ini. 
not write. 

> (Note, however, the version is 0.0.2, and it hasn't been updated with the latest [upstream releases](https://github.com/benhoyt/inih), so I don't know how future-proof it is.)

ah, OK.. @leeebo might be easily able to update it.

leeebo, 2024-07-30:
The upstream [inih](https://github.com/benhoyt/inih) only support parsing `.ini` file, I don't think it will offer write support.

This requirement comes from the [ esp-box ](https://github.com/espressif/esp-box/tree/master/examples/chatgpt_demo)project, We chose `ini`  to better interact with the PC. The esp-box as USB flash drive, and there is a virtual `ini` file in it. Users can modify the `ini` file through PC editor, and then esp-box parses and uses these configuration items. 

@diplfranzhoepfinger any `write` use cases from your side?

diplfranzhoepfinger, 2024-07-30:
> The upstream [inih](https://github.com/benhoyt/inih) only support parsing `.ini` file, I don't think it will offer write support.

no, it does not. 

> @diplfranzhoepfinger any `write` use cases from your side?

yes, and i was able to get [simpleini](https://github.com/brofield/simpleini) running on a ESP32S3. 

it does already write ini Files. 
and read of course.

diplfranzhoepfinger, 2024-07-30:
just found that in the GTK Project there is a alternate Lib for INI Writing and reading: 

https://docs.gtk.org/glib/struct.KeyFile.html

but maybe it is less portable and more Memory intensive ? 

Thanks, 
Franz

diplfranzhoepfinger, 2024-07-30:
@igrr 
@leeebo 

i made you a Demo here: 
https://github.com/DiplomInformatikerFranzHoepfinger/SimpleIni_on_esp32

i uses the https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/file_serving
to enable a easy access to the ini File. 

it writes the followings keys: 
```
[test-Sec]
test-key = ffffffffffffffff
test-key3 = fefefefefefefefe
```

Thanks go out to Peter Wegscheider for contributing its Wrapper (settingsIni.cpp) under the MIT License. 

Regards, 
Franz