espressif/idf-extra-components #362: Channel not in init state

michaelboeding, 2024-08-12:
### Answers checklist.

- [X] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [X] I am using target and esp-idf version as defined in component's idf_component.yml
- [X] I have searched the [issue tracker](https://github.com/espressif/idf-extra-components/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

led_strip

### ESP-IDF version.

5.2

### Development Kit.

Custom Board

### Used Component version.

2.5.4

### More Information.

I'm randomly getting the below errors while using the library. They are not consistent and happen at different times during operation.

E (2460) rmt: rmt_tx_enable(717): channel not in init state
E (2460) led_strip_rmt: led_strip_rmt_refresh(72): enable RMT channel failed

suda-morris, 2024-08-12:
@michaelboeding  Can you check if you're calling `led_strip_refresh` or `led_strip_clear` in different threads without serializing the access to the same led_strip handle?

suda-morris, 2024-08-12:
yes, a mutex is needed.

michaelboeding, 2024-08-12:
I got this implemented and it seems to be working. Thanks for the quick response! I'm going to go ahead and close this. If I see it again ill reopen

michaelboeding, 2024-08-12:
Well maybe I spoke too soon. I'm still seeing this even with mutexs in place on the refresh and clear methods.

W (144550) LED_OBJECT: RMT channel not in init state. Skipping refresh...
E (144590) rmt: rmt_tx_enable(717): channel not in init state
E (144590) led_strip_rmt: led_strip_rmt_refresh(72): enable RMT channel failed
W (144590) LED_OBJECT: RMT channel not in init state. Skipping refresh...
E (144630) rmt: rmt_tx_enable(717): channel not in init state
E (144630) led_strip_rmt: led_strip_rmt_refresh(72): enable RMT channel failed
W (144630) LED_OBJECT: RMT channel not in init state. Skipping refresh...
E (144660) rmt: rmt_tx_enable(717): channel not in init state
E (144660) led_strip_rmt: led_strip_rmt_refresh(72): enable RMT channel failed
W (144660) LED_OBJECT: RMT channel not in init state. Skipping refresh...
E (144690) rmt: rmt_tx_enable(717): channel not in init state
E (144690) led_strip_rmt: led_strip_rmt_refresh(72): enable RMT channel failed
W (144690) LED_OBJECT: RMT channel not in init state. Skipping refresh...
E (144720) rmt: rmt_tx_enable(717): channel not in init state
E (144720) led_strip_rmt: led_strip_rmt_refresh(72): enable RMT channel failed

michaelboeding, 2024-08-13:
And just as a reference heres my code to create a breath effect. Not sure but maybe this has something to do with it ?

```c++

void LED::breath(uint16_t hue, uint8_t saturation) {
    const uint8_t maxValue = 255; // Maximum brightness level
    const uint8_t minValue = 0; // Minimum brightness level
    const int period = 2000; // Breathing period in milliseconds (2 seconds)
    const TickType_t stepDelay = pdMS_TO_TICKS(20); // Delay for each step
    const int steps = period / 20; // Number of steps in one breathing cycle

    for (int i = 0; i < steps; i++) {
        // Check if the mode has changed during the breathing cycle
        if (this->currentMode != LEDMode::CHARGING && this->currentMode != LEDMode::CHARGED && this->currentMode != LEDMode::UNPROVISIONED) {
            // Exit the breathing effect if the mode has changed
            //if we have the mutex then we can unlock it
            return;
        }
        // Calculate the sine wave value for smooth breathing effect
        float sineValue = (std::sin(2.0f * M_PI * i / steps) + 1.0f) / 2.0f;
        uint8_t value = minValue + static_cast<uint8_t>(sineValue * (maxValue - minValue));

        // Set the LED color with the calculated brightness (value)
        this->setPixelColorHSV(hue, saturation, value);

        // Delay for the next step
        vTaskDelay(stepDelay);
    }
}

```

finalyards, 2026-08-29:
@michaelboeding Is this still a concern with version 3 of the component?