espressif/idf-extra-components #375: spi_nand_flash with standard spi mode

txf-, 2024-09-11:
### Is your feature request related to a problem?

I have noticed that the SPI nand flash driver is configured for half duplex via quad spi.

### Describe the solution you'd like.

I'd like the driver to support transactions via standard SPI. Is that possible? 

### Describe alternatives you've considered.

I've tried modifying the example to use standard SPI, but there are warnings about difference in rx_buffer size vs tx_buffer.

igrr, 2024-09-27:
@txf- Sorry for the late response. I think it should be possible to use the standard SPI mode, but the part of the driver which sends SPI transactions needs to be modified. Currently the "write" (MOSI) and "read" (MISO) buffers are provided to spi_nand_execute_transaction function separately. In full duplex (standard SPI) mode, the lengths of these buffers should be the same. We would need to change the API of spi_nand_execute_transaction somewhat to allow for this. Alternatively, spi_nand_execute_transaction could allocate a temporary buffer of `MAX(rx_len, tx_len)` bytes and use it for the transaction, but that would result in higher heap memory usage. We also need to add a configuration field in `spi_nand_flash_config_t` so that the NAND driver knows whether the SPI bus is set up as full or half duplex, and change `spi_nand_*` functions to take the whole `spi_nand_flash_device_t` instead of just the SPI handle.

txf-, 2024-09-27:
> @txf- Sorry for the late response.

It's no problem, I have since looked in the library. I was under the impression that it was half duplex because the Dual and Quad spi modes needed it and I was only using  it in normal spi mode. 

But since then I noticed that this driver doesn't actually use the dual/quad spi commands such as "Fast Read Quad Output (6Bh)" etc. I might be using dual or quad in the future.

Is there any barrier to changing the driver to use these commands?

igrr, 2024-09-27:
No, no specific barrier. The code of this component has mostly been contributed in https://github.com/espressif/esp-idf/pull/9422 and we have used it largely "as is". Based on the open issues in this repo, it seems it is getting more use than we have expected, so we will try to provide more support from our side in the future.

Regarding QIO mode, I don't have a lot of experience with NAND flash, but for SPI NOR flash there is sometimes a difference between vendors in enabling the QIO mode. I guess that might need to be accounted for in the driver if those modes are used. If you end up implementing the commands specific to dual/quad modes, a PR or a patch would be welcome!