espressif/idf-extra-components #379: led_strip has issues with Arduino Framework

zackees, 2024-09-17:
### Answers checklist.

- [X] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [X] I am using target and esp-idf version as defined in component's idf_component.yml
- [X] I have searched the [issue tracker](https://github.com/espressif/idf-extra-components/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

led_strip

### ESP-IDF version.

v5.1 + Arduino Latest

### Development Kit.

ESP32S3 - Xiao S3

### Used Component version.

v2.5.5

### More Information.

Hi there, I'm a core dev for FastLED.

I'm trying to migrate the lib to using the RMT 5.1 w/using the streaming encoder.

## Background

I've been trying to get the RMT 5.1 library to work with our codebase. However, after hitting a wall and double-checking my work I suspect that there is some compatibility issue with the Arduino framework and the `led_strip` component library.

What I did is the following:
  * Ported the led_strip component to C++ (only minimal, necessary changes to get it to compile involving how structs are initialized, I zero memory them then set the members explicitly).
  * Stripped out the idf 4.4 code.
  * Ported a straightforward example (you'll see it https://github.com/fastled/fastled/src/platforms/esp/32/led_strip/rmt_demo.h)
  * Ported a second example right into the dev.ino, just incase I did anything wrong.

It just simply does not work. What's weird is that code runs fine and does not display any errors. However, no LEDs are displayed and the oscilloscope shows no activity on PIN 9.

As a sanity check, I tried out the Arduino RMT 5.1 version and it works flawlessly. Looking through the init libraries I see that there is a lot of initialization that's happening on the pins.

It's also possible that this is my fault somehow. So to check this I've manually diff'd all the source files from our port with the source code in this repo and I can't find anything I'm doing wrong. I've also printed out the states of the various configuration structs and nothing is jumping out at me either. My AI tools also do not identify anything wrong with the port.

Because FastLED is platform neutral and heavily used in esp idf native environments, we can't rely on the Arduino RMT implementation being present, but of course, we must co-exist with it as it is present in >95% of the users using our library.

## Repro case

I've provided a easy repro case so that you can test this yourself. Here is the instructions.

  * Clone https://github.com/FastLED/FastLED
  * Switch to branch `rmt51`
  * Open the FastLED directory in VSCode + PlatformIO
  * Click the Upload button

For the device, you'll need to connect a WS2812 strip to pin 9 of the micro. You should see it blink white and black, while the arduino rmt version cycles a hue. The only significant difference between these two versions is that the Arduino RMT version will write out RGBW data while the idf led_strip version will write out RGB data. But with 9 pixels as a test you'll should still something in either mode.

If you attach an oscilloscope to pin 9 you'll see that expected WS2812 signal being generated for the Arduino RMT impl. However, you will not see any signal other than noise for the idf led_strip version.

This is the output log of running the `led_strip` version:

```
--- Terminal on COM7 | 9600 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
[   127][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Dein[  3706][I][dev.ino:38] setup(): [dev.ino] Start blinking LED strip
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   395028 B ( 385.8 KB)
  Free Bytes        :   361280 B ( 352.8 KB)
  Allocated Bytes   :    28556 B (  27.9 KB)
  Minimum Free Bytes:   356140 B ( 347.8 KB)
  Largest Free Block:   327668 B ( 320.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
    43 : UART_TX[0]
    44 : UART_RX[0]
============ After Setup End =============
[  4430][I][dev.ino:170] led_component_loop(): [dev.ino] Starting loop
[  4505][I][dev.ino:71] configure_led(): [dev.ino] strip_config.strip_gpio_num: 9
[  4591][I][dev.ino:73] configure_led(): [dev.ino] strip_config.max_leds: 9
[  4672][I][dev.ino:74] configure_led(): [dev.ino] strip_config.led_pixel_format: 0
[  4760][I][dev.ino:76] configure_led(): [dev.ino] strip_config.led_model: 0
[  4842][I][dev.ino:77] configure_led(): [dev.ino] strip_config.flags.invert_out: 0
[  4930][I][dev.ino:87] configure_led(): [dev.ino] rmt_config.clk_src: 4
[  5007][I][dev.ino:88] configure_led(): [dev.ino] rmt_config.resolution_hz: 10000000
[  5098][I][dev.ino:89] configure_led(): [dev.ino] rmt_config.mem_block_symbols: 64
[  5187][I][dev.ino:91] configure_led(): [dev.ino] rmt_config.flags.with_dma: 0
[  5271][I][dev.ino:97] configure_led(): [dev.ino] Created LED strip object with RMT backend
[  5369][I][dev.ino:175] led_component_loop(): [dev.ino] Looping
[  5438][I][dev.ino:192] led_component_loop(): [dev.ino] LED OFF!
[  6008][I][dev.ino:175] led_component_loop(): [dev.ino] Looping
[  6077][I][dev.ino:187] led_component_loop(): [dev.ino] LED ON!
[  6654][I][dev.ino:175] led_component_loop(): [dev.ino] Looping
[  6723][I][dev.ino:192] led_component_loop(): [dev.ino] LED OFF!
[  7293][I][dev.ino:175] led_component_loop(): [dev.ino] Looping
```

If we can get this nailed down to the cause of the failure then I will be able to get this new RMT library in FastLED. Yes we will properly namespace it so that it doesn't conflict esp idf users who also include the `led_strip` component library along with FastLED.

Thank you for your consideration and I hope to hear back from you on this. Many users are experiencing boot crash loop and so far I don't have an answer on when or if the RMT 5.1 driver will be able to go in for FastLED.

@me-no-dev

zackees, 2024-09-18:
Just wanted to mention that an easier first step to this issue would be for espressif to make sure that their led_strip component can be compiled in C++.

The portability issue is that the structs use C-specific initialization, and their goto's skip over unallocated stack variables. This is illegal in C++ and there's no #pragma that one can use to get around this.

If `led_strip` is made compatible with C++ then I can just do a straight copy and re-test. If this still doesn't work then we know it's an issue between espressifs Arduino core and their led_strip code or something else. If it does work then this means it's a porting error on my end.

The benefit here is that your users likely want a C++ compatible `led_strip` component as well.

suda-morris, 2024-09-18:
Yes, I can compile a C++ project with the led_strip component. Here is the initialization code:

```c++
#include <stdio.h>
#include "led_strip.h"

#define BLINK_GPIO 0

extern "C" void app_main(void)
{
    led_strip_handle_t led_strip;

    /* LED strip initialization with the GPIO and pixels number*/
    led_strip_config_t strip_config = {
        .strip_gpio_num = BLINK_GPIO, // The GPIO that connected to the LED strip's data line
        .max_leds = 1, // The number of LEDs in the strip,
        .led_pixel_format = LED_PIXEL_FORMAT_GRB, // Pixel format of your LED strip
        .led_model = LED_MODEL_WS2812, // LED strip model
        .flags = {
            .invert_out = false,
        }
    };

    led_strip_rmt_config_t rmt_config = {
        .clk_src = RMT_CLK_SRC_DEFAULT, // different clock source can lead to different power consumption
        .resolution_hz = 10 * 1000 * 1000, // 10MHz
        .mem_block_symbols = 0, // 0 means use the default value
        .flags = {
            .with_dma = false,
        }
    };
    ESP_ERROR_CHECK(led_strip_new_rmt_device(&strip_config, &rmt_config, &led_strip));

    led_strip_spi_config_t spi_config = {
        .clk_src = SPI_CLK_SRC_DEFAULT, // different clock source can lead to different power consumption
        .spi_bus = SPI2_HOST,   // SPI bus ID
        .flags = {
            .with_dma = false,
        }
    };
    ESP_ERROR_CHECK(led_strip_new_spi_device(&strip_config, &spi_config, &led_strip));
}
```

zackees, 2024-09-18:
This is great. I the curly braces for the anon structs is what I was looking for. TIL.

I'm going to retry the port now with this cleaner syntax.

However, you are missing the second part of this portability problem - which is the fact that C++ does not allow goto across unallocated variables.

Like this

```C
esp_err_t led_strip_new_rmt_device(const led_strip_config_t *led_config, const led_strip_rmt_config_t *rmt_config, led_strip_handle_t *ret_strip)
{
    led_strip_rmt_obj *rmt_strip = NULL;
    esp_err_t ret = ESP_OK;
    ESP_GOTO_ON_FALSE(led_config && rmt_config && ret_strip, ESP_ERR_INVALID_ARG, err, TAG, "invalid argument");
    ESP_GOTO_ON_FALSE(led_config->led_pixel_format < LED_PIXEL_FORMAT_INVALID, ESP_ERR_INVALID_ARG, err, TAG, "invalid led_pixel_format");
    uint8_t bytes_per_pixel = 3;
    if (led_config->led_pixel_format == LED_PIXEL_FORMAT_GRBW) {
        bytes_per_pixel = 4;
    } else if (led_config->led_pixel_format == LED_PIXEL_FORMAT_GRB) {
        bytes_per_pixel = 3;
    } else {
        assert(false);
    }
    rmt_strip = calloc(1, sizeof(led_strip_rmt_obj) + led_config->max_leds * bytes_per_pixel);
    ESP_GOTO_ON_FALSE(rmt_strip, ESP_ERR_NO_MEM, err, TAG, "no mem for rmt strip");
    uint32_t resolution = rmt_config->resolution_hz ? rmt_config->resolution_hz : LED_STRIP_RMT_DEFAULT_RESOLUTION;

    // for backward compatibility, if the user does not set the clk_src, use the default value
    rmt_clock_source_t clk_src = RMT_CLK_SRC_DEFAULT;
    if (rmt_config->clk_src) {
        clk_src = rmt_config->clk_src;
    }
    size_t mem_block_symbols = LED_STRIP_RMT_DEFAULT_MEM_BLOCK_SYMBOLS;
    // override the default value if the user sets it
    if (rmt_config->mem_block_symbols) {
        mem_block_symbols = rmt_config->mem_block_symbols;
    }
    rmt_tx_channel_config_t rmt_chan_config = {
        .clk_src = clk_src,
        .gpio_num = led_config->strip_gpio_num,
        .mem_block_symbols = mem_block_symbols,
        .resolution_hz = resolution,
        .trans_queue_depth = LED_STRIP_RMT_DEFAULT_TRANS_QUEUE_SIZE,
        .flags.with_dma = rmt_config->flags.with_dma,
        .flags.invert_out = led_config->flags.invert_out,
    };
    ESP_GOTO_ON_ERROR(rmt_new_tx_channel(&rmt_chan_config, &rmt_strip->rmt_chan), err, TAG, "create RMT TX channel failed");

    led_strip_encoder_config_t strip_encoder_conf = {
        .resolution = resolution,
        .led_model = led_config->led_model
    };
    ESP_GOTO_ON_ERROR(rmt_new_led_strip_encoder(&strip_encoder_conf, &rmt_strip->strip_encoder), err, TAG, "create LED strip encoder failed");

    rmt_strip->bytes_per_pixel = bytes_per_pixel;
    rmt_strip->strip_len = led_config->max_leds;
    rmt_strip->base.set_pixel = led_strip_rmt_set_pixel;
    rmt_strip->base.set_pixel_rgbw = led_strip_rmt_set_pixel_rgbw;
    rmt_strip->base.refresh = led_strip_rmt_refresh;
    rmt_strip->base.clear = led_strip_rmt_clear;
    rmt_strip->base.del = led_strip_rmt_del;

    *ret_strip = &rmt_strip->base;
    return ESP_OK;
err:
    std::cout << "Error happened" << std::endl;
    if (rmt_strip) {
        if (rmt_strip->rmt_chan) {
            rmt_del_channel(rmt_strip->rmt_chan);
        }
        if (rmt_strip->strip_encoder) {
            rmt_del_encoder(rmt_strip->strip_encoder);
        }
        free(rmt_strip);
    }
    return ret;
}
```

zackees, 2024-09-18:
Also OR-ing enums is not allowed in C:

```
state |= RMT_ENCODING_COMPLETE;
```

zackees, 2024-09-18:
@suda-morris 

Thanks, go it blinking. Something was wrong with my port.

closed 2024-09-18 by zackees

zackees, 2024-09-18:
I ported the led_strip to C++ and made an open source repo anyone can use. Minimal dependencies - only esp headers are used. Compatible with Arduino and platform io.

The IDF 4.4 and SPI code was stripped out.

https://github.com/zackees/esp-rmt-led-strip-component-idf-5-1-cpp