espressif/idf-extra-components #401: esp_lcd_qemu_rgb add esp32s3 support

zhengzhu-1m, 2024-10-10:
### Is your feature request related to a problem?

- https://github.com/espressif/qemu/releases/tag/esp-develop-9.0.0-20240606
```
I (2142) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
I (2143) boot: compile time Oct 10 2024 23:12:24
I (2144) boot: Multicore bootloader
I (2153) boot: chip revision: v0.3
I (2158) qio_mode: Enabling QIO for flash chip ISSI
I (2162) boot.esp32s3: Boot SPI Speed : 80MHz
I (2162) boot.esp32s3: SPI Mode       : QIO
I (2163) boot.esp32s3: SPI Flash Size : 16MB
I (2167) boot: Enabling RNG early entropy source...
I (2173) boot: Partition Table:
I (2173) boot: ## Label            Usage          Type ST Offset   Length
I (2174) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (2175) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (2176) boot:  2 factory          factory app      00 00 00010000 00700000
I (2180) boot: End of partition table
I (2183) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=4f0ech (323820) map
I (2391) esp_image: segment 1: paddr=0005f114 vaddr=3fc95c00 size=00f04h (  3844) load
I (2401) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=49570h (300400) map
I (2584) esp_image: segment 3: paddr=000a9598 vaddr=3fc96b04 size=02180h (  8576) load
I (2593) esp_image: segment 4: paddr=000ab720 vaddr=40374000 size=11bech ( 72684) load
I (2693) boot: Loaded app from partition at offset 0x10000
I (2695) boot: Disabling RNG early entropy source...
I (2707) cpu_start: Multicore app
I (2720) cpu_start: Pro cpu start user code
I (2720) cpu_start: cpu freq: 240000000 Hz
I (2720) app_init: Application information:
I (2720) app_init: Project name:     lvgl-jc3248w535
I (2720) app_init: App version:      1
I (2720) app_init: Compile time:     Oct 10 2024 23:11:06
I (2720) app_init: ELF file SHA256:  eb52c3863...
I (2721) app_init: ESP-IDF:          v5.3.1-dirty
I (2721) efuse_init: Min chip rev:     v0.0
I (2722) efuse_init: Max chip rev:     v0.99 
I (2722) efuse_init: Chip rev:         v0.3
I (2723) heap_init: Initializing. RAM available for dynamic allocation:
I (2723) heap_init: At 3FCA9848 len 0003FEC8 (255 KiB): RAM
I (2723) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (2723) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (2723) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (2728) spi_flash: detected chip: issi
I (2728) spi_flash: flash io: qio
W (2728) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (2729) sleep: Configure to isolate all GPIO pins in sleep state
I (2731) sleep: Enable automatic switching of GPIO sleep configuration
I (2740) main_task: Started on CPU0
I (2746) main_task: Calling app_main()
I (2750) LVGL: Starting LVGL task
I (2752) lcd_axs15231b: Install RGB LCD panel driver
// error
E (2757) lcd_qemu.rgb: esp_lcd_new_rgb_qemu(56): qemu panel is not available on real hardware
E (2759) lcd_axs15231b: lcd_display_new(530): New panel failed
ESP_ERROR_CHECK failed: esp_err_t 0x106 (ESP_ERR_NOT_SUPPORTED) at 0x420094b1
```

### Describe the solution you'd like.

esp qemu Added ESP32-S3 support
- https://github.com/espressif/qemu/releases/tag/esp-develop-9.0.0-20240606
Hopefully esp_lcd_qemu_rgb compatible

igrr, 2024-10-11:
Hi @zhengzhu-1m, [esp-develop-9.0.0-20240606](https://github.com/espressif/qemu/releases/tag/esp-develop-9.0.0-20240606) release of QEMU doesn't include the virtual framebuffer peripheral for ESP32-S3, which is why you get this error. We'll add this peripheral in the next QEMU release.

closed 2024-10-12 by zhengzhu-1m

o-marshmallow, 2024-11-13:
The peripheral has been implemented for ESP32-S3 target in QEMU (commit https://github.com/espressif/qemu/commit/6f41181223a0c8c01571c908045aa5b6ab8f2e89). It is already available in our development branch, and will be available in the next release of QEMU.