espressif/idf-extra-components #405: dhara

anastasia-be, 2024-10-14:


igrr, 2024-10-15:
Currently dhara component is tested by spi_nand_flash component examples/tests.

Since dhara "port layer" is inside "spi_nand_flash", it's a bit difficult to write an example for dhara in isolation. We probably need to refactor spi_nand_flash to make this possible.

anastasia-be, 2024-10-16:
@igrr thank you for your feedback! 

Let’s coordinate on finalizing the plan for this work and I will set a task for refactor of the spi_nand_flash to separate out the dhara component.

omnitex, 2026-10-22:
Dhara is integrated in spi_nand_flash and provides only the FTL logic, requiring a driver to actually do anything, it's not a standalone component.

Existing spi_nand_flash test app and host test are sufficient.

closed 2026-10-22 by omnitex