espressif/idf-extra-components #424: Error: "Failed to resolve component 'spi_nand_flash'

Teesmo, 2024-10-16:
### Answers checklist.

- [X] I have read the documentation of the component in question and the issue is not addressed there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I downloaded the archive: espressif__spi_nand_flash from the component registry, When I tried running the example provided in the package., I encountered the error : `Failed to resolve component  'spi_nand_flash'`. I am not sure how to proceed further, because I was under the impression that the example would run with no interventions on my side. 

Here is the call stack:
`Processing 3 dependencies:`
`[3/3] idf (5.4.0)CMake Error at /home/onepower/esp/esp-idf/tools/cmake/build.cmake:268 (message): `
  `Failed to resolve component 'spi_nand_flash'.`
`Call Stack (most recent call first): `
  `/home/onepower/esp/esp-idf/tools/cmake/build.cmake:310 (__build_resolve_and_add_req) `
  `/home/onepower/esp/esp-idf/tools/cmake/build.cmake:607 (__build_expand_requirements) `
  `/home/onepower/esp/esp-idf/tools/cmake/project.cmake:710 (idf_build_process) `
  `CMakeLists.txt:8 (project)`
`

igrr, 2024-10-16:
Hi @Teesmo,

If you download the archive from the component registry manually and want to run the example from the archive, make sure you extract the archive into a directory called `spi_nand_flash` or `espressif__spi_nand_flash`. Then the example will work.
```
mkdir spi_nand_flash
cd spi_nand_flash
unzip path/to/component_zip_file
cd examples/nand_flash
idf.py build
```

Alternatively, you can skip downloading the archive manually, and use this command to get the example project directly:
```
idf.py create-project-from-example "espressif/spi_nand_flash^0.4.0:nand_flash"
cd nand_flash
idf.py build
```
The latter is the recommended way to run examples of components from the component registry.

Teesmo, 2024-10-16:
Let me try that, thank you.

Teesmo, 2024-10-16:
It built, thank you

igrr, 2024-10-16:
@Teesmo Out of curiosity, which NAND flash chip are you using?

Teesmo, 2024-10-16:
I am using the MT29F1G01ABAFDSF-AAT:F. I was about to open an issue explaining that my chip is not being detected at all (The chip restarts after the error ` nand: spi_nand_flash_init_device(233): Failed to detect nand chip`.
I have edited the nand.c  and nand_devices.h files to cater for my chip :
The changes I have added to nand.c are:
`case MICRON_DI_34:`
        `dev->read_page_delay_us = 115;`
        `dev->erase_block_delay_us = 2000;`
        `dev->program_page_delay_us = 240;`
        `dev->dhara_nand.num_blocks = 2048;`
        `dev->dhara_nand.log2_ppb = 6;      // 64 pages per block`
        `dev->dhara_nand.log2_page_size = 12; // 4096 bytes per page`
        `break;`
    `case MICRON_DI_14:`
        `dev->read_page_delay_us = 46;`
        `dev->erase_block_delay_us = 2000;`
        `dev->program_page_delay_us = 220;`
        `dev->dhara_nand.num_blocks = 1024;`
        `dev->dhara_nand.log2_ppb = 6;     // 64 pages per block`
        `dev->dhara_nand.log2_page_size = 11; // 2048 bytes per page`
        `break;`
`case MICRON_DI_15:`
        `dev->read_page_delay_us = 46;`
        `dev->erase_block_delay_us = 2000;`
        `dev->program_page_delay_us = 220;`
        `dev->dhara_nand.num_blocks = 1024;`
        `dev->dhara_nand.log2_ppb = 6;      // 64 pages per block`
        `dev->dhara_nand.log2_page_size = 11; // 2048 bytes per page`
        `break;`
I have also added the device ID macros to nand_devices.h.

igrr, 2024-10-16:
Do I understand it right that with these changes, your chip is detected and the example works?

If yes, would you mind opening a [pull request](https://github.com/espressif/idf-extra-components/pulls) with your changes?

Teesmo, 2024-10-16:
No. The issue is that with or without my changes, my chip is not being detected (for some reason). I think my changes, under normal circumstances, should work. I checked if my issue is a hardware problem, but my wiring seems OK, and my GPIO allocations are inline with the example provided.

igrr, 2024-10-16:
> I understood that the author had run the code and it had (sort of) worked, even before making changes that are specific to his chip.

No, the basic changes (like adding chip ID) are still necessary, otherwise the chip won't be detected.

I don't have this specific chip (just got the `MT29F2G` variant...), but you can try to debug on your side by adding logging of vendor ID and chip ID. If you get something like `0x00` or `0xff` back, this means that something is wrong with the wiring or the chip is not working.

Teesmo, 2024-10-16:
I tried logging and got 0xff off two chips already. I'll double check my wiring.

igrr, 2024-10-16:
> Apart from the basics (chip ID etc), will the extra changes (page_delay, block_delay etc) affect chip detection?

No, as you can see from the code, we just get the VID and PID, and then there are switch-case statements to select by VID and PID. All the parameters set for the specific chip (page_delay etc) are only used once we exit from the chip detection code.

The only thing necessary for chip detection to work is to have MOSI, MISO, CLK, CS lines correctly connected, and the flash chip should have power supply.

Teesmo, 2024-10-16:
Maybe my pin mappings might be wrong. I'll check that as well.

Teesmo, 2024-10-16:
I've checked my pin allocations, and they seem correct. I'm wondering, maybe this (NAND) does not work on my microcontroller (ESP32-S2). I'll send the pull request

igrr, 2024-10-16:
0xff means either of the following things:

- flash chip is not receiving the clock signal from ESP
- flash chip is not receiving CS signal from ESP
- ESP is not receiving the MISO signal from the flash chip

If you have an oscilloscope or a logic analyzer you can probe the signals on both sides and try to see if all the lines are properly connected. You can also share which pins you use, I can try to run an example on esp32-s2, although with a slightly different flash model.

Teesmo, 2024-10-16:
I use the default GPIO pins as defined in soc/spi_pins.h (9 HD, 10 CS, 11 MOSI, 12 CLK, 13 MISO, 14 WP). Although I think my implementation does not use HD and WP.

RathiSonika, 2025-05-22:
Hello @Teesmo ,
Based on the pin numbers you mentioned, the spi_nand_flash component should work with the ESP32-S2.
Have you been able to get the code working? If so, feel free to open a PR for MT29F1G01 support. If not, let me know if you need any help with this.
Thank you!

Teesmo, 2025-05-22:
Hi @RathiSonika, we had [this pull request ](https://github.com/espressif/idf-extra-components/pull/432) approved some months ago.

RathiSonika, 2025-05-22:
Ah, my mistake! The README.md wasn’t updated, so I missed it. Anyway, I will update it. Thanks for the clarification!
Closing this issue as it is already done.
Thank you!

closed 2025-05-22 by RathiSonika