espressif/idf-extra-components #438: led_strip commit 9a664de broke down esp_bsp_generic builds

rishi9699, 2024-10-26:
### Answers checklist.

- [X] I have read the documentation of the component in question and the issue is not addressed there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

This line 44 in the file led_strip/include/led_strip_types.h was removed,
`    led_pixel_format_t led_pixel_format; /*!< LED pixel format */
`
This has caused esp_bsp_generic builds to fail. I tried running the 'light' example in esp-matter and the build is failing. Giving me the following log,

> /home/rishi/light/managed_components/espressif__esp_bsp_generic/src/esp_bsp_generic.c:197:6: error: 'led_strip_config_t' has no member named 'led_pixel_format'
>   197 |     .led_pixel_format = LED_PIXEL_FORMAT_GRB, // Pixel format of your LED strip
>       |      ^~~~~~~~~~~~~~~~
> /home/rishi/light/managed_components/espressif__esp_bsp_generic/src/esp_bsp_generic.c:197:25: error: 'LED_PIXEL_FORMAT_GRB' undeclared here (not in a function)
>   197 |     .led_pixel_format = LED_PIXEL_FORMAT_GRB, // Pixel format of your LED strip
>       |                         ^~~~~~~~~~~~~~~~~~~~
> /home/rishi/light/managed_components/espressif__esp_bsp_generic/src/esp_bsp_generic.c:222:22: error: initializer element is not constant
>   222 |     .led_strip_cfg = bsp_leds_rgb_strip_config,

rishi9699, 2024-10-27:
I have created a [pull request](https://github.com/espressif/esp-bsp/pull/423) for fixing the same.

igrr, 2024-10-27:
Thank tou for opening the issue and making a PR! The issue in esp-bsp is probably sufficient, since the issue is going to be fixed there. I will close this one.

closed 2024-10-27 by igrr