espressif/idf-extra-components #449: libsodium component is missing sources from CMakeLists

bm-l, 2024-11-28:
### Answers checklist.

- [X] I have read the documentation of the component in question and the issue is not addressed there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

A handful of upstream source files are missing from the CMakeLists.txt provided for libsodium.

Among them:

- aead_aegis256.c
- aegis256_soft.c
- aead_aegis128l.c
- aegis128l_soft.c
- softaes.c

This results in build failures with "undefined reference to..." when using the libsodium component.

mahavirj, 2024-11-28:
@bm-l  Please try your build with https://github.com/espressif/idf-extra-components/pull/450

bm-l, 2024-11-28:
Build was successful, thankyou 😃

closed 2024-11-29 by mahavirj