espressif/idf-extra-components #464: Question regarding ESP32P4 and OpenCV component

Ncouch64, 2025-01-06:
### Answers checklist.

- [X] I have read the documentation of the component in question and the issue is not addressed there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello I apologize I am making another issue post but I am at a loss when it comes to the new P4 board. I thought I had my project working on the board before realizing there is a different driver for working with the cam now because the dev board is using a mipi camera. Before when I thought I was capturing images it was not functioning.

I am wondering if there are any examples available that show something as simple as capturing a single image. I went to the esp_cam_sensor component but the examples there seem to be a little complex. I believe esp32-cam component will not work with this dev board due to the mipi connection. Any recommendations?

suda-morris, 2025-01-07:
@Ncouch64  Yes, there's one MIPI CSI Camera example in the esp-idf: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/camera/mipi_isp_dsi

Ncouch64, 2025-01-07:
> @Ncouch64 Yes, there's one MIPI CSI Camera example in the esp-idf: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/camera/mipi_isp_dsi

Thank you! Will take a look.

closed 2025-01-08 by Ncouch64