espressif/idf-extra-components #473: Request for upstream merge of FastLED's improvements to espressif's led_strip

zackees, 2025-02-03:
### Is your feature request related to a problem?

Hi there, core dev for FastLED.

We made a significant enhancement to the led_strip library from espressif.

We want to merge some of the successes back into espressif. Also, I'm going to mention some of our failures with more advanced features we tried to bolt on.

## Two most significant features

*these are both ready to integrate now*

  * async draw. Both RMT and SPI. This is extremely easy to implement and your code base was almost already there. This produces MASSIVE performance gains as the next frame is free to be computed while the current frame hardware bitbangs out.
  * exposing led timing T0H...T1L
    * allows support for any other clockless chipsets like WS2816, WS2815, etc... to be represented by your strip.
    * Instead of passing an enum for the LED strip type, the user passes in the encoder timings explicitly.

## Abandoned feature: Moving RMT controllers from one GPIO to another, enabling RMT-worker pools.

  * This is missing in RMT5, but was present in RMT4
  * We originally did this ourselves through eager teardown after render and RMT reinstatation on next draw. This would allow us to preserve our RMT-workerpool when number of strips > built in RMT units.
    * Our design featured
      * Changing the LED strip array from built into to the RMT controller in the api, to being a pointer.
        * This allowed us to tear down the RMT unit after completion without fragmenting the memory with a variable length embedded pixel array object.
        * What went wrong? After RMT teardown, the GPIO pin on many ESP32-Wroom devices was left floating which caused a transient HIGH pin state with enough amplitude to trigger an errant T1H signal to the LEDs on the next first pulse. It was discovered that the fix was a weak pulldown on the pin, but this was discovered after we abandoned the feature and rebased off of led_strip 3.0.

## Broken Feature: RMT Memory settings
  * We doubled the memory for ESP32-C3 from 48 -> 96 to prevent flicker during WIFI. According to the datasheet there is 192 RMT symbols that is shared between all RMT devices on -C3. So at 96 RMT words per RMT device, we should have the usage of 2 strips with flicker free LED output during WIFI. However, according to this [bug report](https://github.com/FastLED/FastLED/issues/1873) only one is available before the led_strip API issues a `readyE (1104) rmt: rmt_tx_register_to_group(152): no free tx channels`. It's entirely possible the user happens to be using another RMT for something else or a WS2812 led status light is being implicitly used. I'm following up with this.

## Maybe useless feature: RMT DMA

  * We only get one DMA channel on RMT. The docs don't mention this. We have never been able to get this to work in our codebase. Although we suspect this is our issue as  https://github.com/suda-morris was able to help us prove with a unit test that this does indeed work. I've submitted a [PR](https://github.com/espressif/idf-extra-components/pull/467) to include this in your unit tests, but espressif has yet to review this PR.
    * We were really hopeful that RMT-DMA was going to be the answer. We aren't entirely sure what the specifics are for why this can't be supported on more than one RMT channel. There is no documentation on whether this allows more memory to be allocated to the non-rmt strips. We'd have to experiment on our side to see if this works, but we have an issue that we need to test this across the DMA supporting chipset family, however, this may only actually be the ESP32-S3 chipsets, as non of the other chipsets seem to support DMA.

## Broken feature: SPI leds during WIFI transmission

  * A bug has been filed on this [issue](https://github.com/espressif/idf-extra-components/issues/334) by others.

## Broken Feature: RMT on ESP32-S3 loses 3 RMT units instead of 2 when doubling the memory.

  * This is to reduce flicker with wifi. We expected that doubling the memory would reduce the available RMT workers by half (-2) but instead it results in losing three workers. I'm having trouble finding our bug reports on this, so my apologies.

## Missing Feature: I2S Based Led Strip

This is a big feature and a third party (Yves) already implemented this which was integrated into FastLED. We have versions for ESP32dev and ESP32-S3 for IDF 5.1+. This driver is serving as our "escape hatch" for users not being able to drive enough WS2812 leds in parallel. How exactly it works is a bit mysterious for me but could be alien technology (just kidding). It somehow works by transposing leds into a format the DMA channel expects. We can drive up to 16 leds strips with ESP32-S3, and 24 strips with ESP32dev in IDF4.x. I've recently made the ESP32dev I2S driver compiler compatible with IDF5.x, but I have no idea if it works.

The most significant one being that we change the led_strip API so that it supports async drawing.

## Summary

Two very significant features can be implemented right away: async draw and led chipset timings. These are production ready and extremely well tested.

I2S demonstrations are available. It's in C++ though and would need to be re-written for espressifs purpose. But their ability to drive an insane number of LEDs has been proven to be extremely valuable.

For RMT the biggest challenge for us is being able to move RMT workers from one GPIO to another. It would be really awesome if this can be done without an IDF core change.

Other bugs with RMT and SPI were presented. Not sure what the issue is or what the fix is.

## Links

  * FastLED's enhanced led_strip component library: 
    * https://github.com/FastLED/FastLED/tree/master/src/third_party/espressif/led_strip/src
  * Yves I2S driver
    * ESP32dev: https://github.com/FastLED/FastLED/blob/12f8e82d64b445217694362f793fea5f6b93f1fe/src/platforms/esp/32/i2s/i2s_esp32dev.h#L1
    * Esp32-S3: https://github.com/FastLED/FastLED/tree/master/src/third_party/yves/I2SClockLessLedDriveresp32s3

Thanks for your attention:

~Zach Vorhies - core maintainer for [FastLED](https://github.com/fastled/fastled).

Kainarx, 2025-03-14:
Thank you for your detailed analysis and improvement proposals! @zackees 

Asynchronous drawing and LED timing customization are both highly valuable features, and we will evaluate how to merge them into the upstream.

### Regarding broken functionalities:

1. Only in ESP32 and ESP32S2, all channels can work as TX or RX. However, in ESP32C3, while it has 4 channels sharing 192 RMT symbols, **channels 0–1 are dedicated to TX direction, and channels 2–3 are dedicated to RX direction.** If you use 96 symbols in channel 0, channel 1 will be occupied, while channels 2–3 do not support the TX direction. As a result, the driver cannot allocate additional TX channels. 

2. For more details on DMA support, you can check our [TRM](https://www.espressif.com.cn/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf).
[image]
**Not all RMT channels support DMA**, as this is determined by hardware and cannot be implemented through software. Besides, **SPI always supports DMA**

3. Since Wi-Fi interrupts have a higher priority than peripherals, in some cases, peripheral interrupts may not be processed in time, leading to timing errors. To mitigate this issue, you can:
**Move the LED strip application to another core (if available) or enable SPI DMA.**

4. The root cause is likely the same as issue 1, so you may need to add an option to balance the number of LED strips and performance.

### Regarding missing functionalities:

1. At specific frequencies, SPI can use "100" to represent logic 0 and "110" to represent logic 1. Alternatively, we can use one byte to represent a logic 0 or 1. This approach sacrifices resolution for performance and memory efficiency. We will further evaluate whether to expand resolution to support more timing formats.
2. We're delighted that third-party developers have contributed an I2S LED driver! However, we currently do not plan to provide official support. Instead, newer chips like C6 and H2 feature a **Parallel IO** peripheral, which allows controlling 8/16 LED strips simultaneously. We are considering writing a driver for this—stay tuned!

zackees, 2025-03-14:
Thank you for this very technical response. I look forward to seeing the parallel io API in the led_strip api and pushing the limit.

If I may ask a question, why does the networking / wifi seem to be so granular in its occupation with the mcu? Are there big chunks of data that are being processed from an isr? This seems to be the root cause.

Is it possible to

  * take the top offender and break up its hold on the mcu and yield to lower priority irs?

  * to provide an asm shim to allow level 7 interrupts bridge to c function callbacks?

As for core 0 or 1, fastled runs in the core that is opposite of the networking core (arduino/platformio)

igrr, 2025-03-14:
> to provide an asm shim to allow level 7 interrupts bridge to c function callbacks?

For chips with Xtensa cores: this is technically possible, even though this requires a bit of work, see e.g. [hli_vectors.S](https://github.com/espressif/esp-idf/blob/master/components/bt/controller/esp32/hli_vectors.S) and [hli_api.c](https://github.com/espressif/esp-idf/blob/master/components/bt/controller/esp32/hli_api.c).

For chips with RISC-V cores: high-level interrupt handlers can be written in C.

However we can't use RTOS primitives (such as queues or spinlocks) from the high-level interrupts.

zackees, 2025-03-14:
Does someone smarter than me know how to write an Xtensa asm shim for a c callback.

zackees, 2025-03-14:
> For chips with Xtensa cores: this is technically possible, even though this requires a bit of work, see e.g. [...]

I'm in the python and C++ world and this seems like trying to read pure jedi.

zackees, 2025-05-03:
yes

zackees, 2025-06-07:
Wanted to follow up. I see that 3.0.1 is released with bit timings for WS2811. Thank you.

Can you please reconsider doing an I2S driver?

This is simply THE protocol to use for the S3 flagship product. It's what power users of FastLED/ESP32-XX are choosing to use.

Resistant to flicker from WIFi because it's all DMA'd, and with a shift register it can drive 120 channels of WS281x. See the virtual pins library by hpwit/Yves.

The challenge right now with enabling the I2S driver in FastLED is the requirement to use PlatformIO in order to use the build setting (COME BACK AND INSERT LINK) to enable PSRAM, which is not the default in Arduino. My recommendation, naive as it may be, is to enable PSRAM by default, if this is possible, for the Arduino platform. This would allow us to enable IS2 everywhere, regardless of whether espressif makes their own I2S WS281x driver.

Jason2866, 2025-06-08:
This settings are **only** needed for old esp32 revisions 
```
-mfix-esp32-psram-cache-issue
-mfix-esp32-psram-cache-strategy=memw
```
NO other MCU does need this bug around settings

zackees, 2025-06-09:
> This settings are **only** needed for old esp32 revisions 
> ```
> -mfix-esp32-psram-cache-issue
> -mfix-esp32-psram-cache-strategy=memw
> ```
> NO other MCU does need this bug around settings

What about partition.csv file?

Can that be a default? Or am I misguided?