espressif/idf-extra-components #475: pre_encrypted_ota question

dizcza, 2025-02-15:
Don't you need to call `esp_encrypted_img_decrypt_abort()` here

https://github.com/espressif/idf-extra-components/blob/4eac7a4de7f8ca3d7798979f34f84bffd4f8090c/esp_encrypted_img/examples/pre_encrypted_ota/main/pre_encrypted_ota.c#L172

?

Another question, does anyone plan to port this encryption functionality to Arduino? Sadly, `CONFIG_ESP_HTTPS_OTA_DECRYPT_CB` is disabled by default so I cannot simply use my ESP-IDF code on Arduino. And like many others, I disagree with the official recommendations posted here https://esp32.com/viewtopic.php?f=13&t=31410 suggesting using Flash encryption only and storing binaries as plain text on servers. Nor do I want to write a workaround to mitigate this issue.

hrushikesh430, 2025-06-16:
Hello @dizcza,

1. For the first question, the issue is already fixed, you can check it on [master](https://github.com/espressif/idf-extra-components/blob/master/esp_encrypted_img/examples/pre_encrypted_ota/main/pre_encrypted_ota.c#L206).
2. In the case of Arduino, the main issue is that we provide pre-built binaries of ESP-IDF, which means ESP-IDF is not configurable and the default configurations are enforced. Typically, in such cases, users require an API or some configuration option to enable features dynamically. Given this situation, I don't see any alternative to using this feature dynamically other than enabling DECRYPT_CB by default.

dizcza, 2025-06-17:
All right. I've switched to Arduino as a component.

closed 2025-06-17 by dizcza