espressif/idf-extra-components #479: onewire fails on idf 5.4

mkellner, 2025-02-28:
### Answers checklist.

- [x] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [x] I am using target and esp-idf version as defined in component's idf_component.yml
- [x] I have searched the [issue tracker](https://github.com/espressif/idf-extra-components/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

onewire_bus

### ESP-IDF version.

v5.4

### Development Kit.

generic nodemcu esp32

### Used Component version.

onewire_bus: 1.0.2, ds18b20: 0.1.2

### More Information.

Using idf v5.3.2, I have success with onwire_bus and ds18b20.

Using idf v5.4, I have failure.

mkellner, 2025-03-01:
```�ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6276
load:0x40078000,len:15716
load:0x40080400,len:4
--- 0x40080400: _init at ??:?

ho 8 tail 4 room 4
load:0x40080404,len:3860
entry 0x4008063c
I (31) boot: ESP-IDF v5.4 2nd stage bootloader
I (31) boot: compile time Feb 28 2025 15:53:11
I (31) boot: Multicore bootloader
I (32) boot: chip revision: v1.0
I (35) boot.esp32: SPI Speed      : 40MHz
I (38) boot.esp32: SPI Mode       : DIO
I (42) boot.esp32: SPI Flash Size : 2MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (78) boot: End of partition table
I (82) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=097bch ( 38844) map
I (102) esp_image: segment 1: paddr=000197e4 vaddr=3ff80000 size=0001ch (    28) load
I (103) esp_image: segment 2: paddr=00019808 vaddr=3ffb0000 size=0254ch (  9548) load
I (110) esp_image: segment 3: paddr=0001bd5c vaddr=40080000 size=042bch ( 17084) load
I (121) esp_image: segment 4: paddr=00020020 vaddr=400d0020 size=14170h ( 82288) map
I (150) esp_image: segment 5: paddr=00034198 vaddr=400842bc size=09460h ( 37984) load
I (171) boot: Loaded app from partition at offset 0x10000
I (171) boot: Disabling RNG early entropy source...
I (182) cpu_start: Multicore app
I (190) cpu_start: Pro cpu start user code
I (190) cpu_start: cpu freq: 160000000 Hz
I (190) app_init: Application information:
I (190) app_init: Project name:     ds18b20-read
I (194) app_init: App version:      3f1937e-dirty
I (199) app_init: Compile time:     Feb 28 2025 15:53:03
I (204) app_init: ELF file SHA256:  33e76adc3...
I (208) app_init: ESP-IDF:          v5.4
I (212) efuse_init: Min chip rev:     v0.0
I (216) efuse_init: Max chip rev:     v3.99 
I (220) efuse_init: Chip rev:         v1.0
I (224) heap_init: Initializing. RAM available for dynamic allocation:
I (230) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (235) heap_init: At 3FFB2DD0 len 0002D230 (180 KiB): DRAM
I (240) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (245) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (251) heap_init: At 4008D71C len 000128E4 (74 KiB): IRAM
I (257) spi_flash: detected chip: generic
I (260) spi_flash: flash io: dio
W (263) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (276) main_task: Started on CPU0
I (286) main_task: Calling app_main()
I (286) DS18B20: Device iterator created, start searching...
E (1286) 1-wire.rmt: onewire_bus_rmt_reset(393): 1-wire reset pulse receive timeout
E (1286) 1-wire.device: onewire_device_iter_get_next(61): reset bus failed
E (1286) rmt: rmt_receive(419): channel not in enable state
E (1286) 1-wire.rmt: onewire_bus_rmt_reset(386): 1-wire reset pulse receive failed
E (1296) 1-wire.device: onewire_device_iter_get_next(61): reset bus failed
E (1306) rmt: rmt_receive(419): channel not in enable state
E (1306) 1-wire.rmt: onewire_bus_rmt_reset(386): 1-wire reset pulse receive failed
```

suda-morris, 2025-03-20:
Hi @mkellner I ran the example on the IDF master branch but see it's working as expected. https://github.com/espressif/esp-idf/tree/master/examples/peripherals/rmt/onewire

```
NOTICE: Processing 3 dependencies:
NOTICE: [1/3] espressif/ds18b20 (0.1.2)
NOTICE: [2/3] espressif/onewire_bus (1.0.2)
NOTICE: [3/3] idf (5.5.0)
```

Can you try that example again?

mkellner, 2025-04-04:
My apologies.
I had a hardware issue.

closed 2025-04-04 by mkellner