espressif/idf-extra-components #483: I can't use esp-clang build idf project (IDFGH-14858)

NewNoob2002, 2025-03-15:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

should same to gcc, can compile successfully.

### What is the actual behavior?

In the first, it will show no __STDC_HOSTED__, i see this is in CMakeLists.txt , and i try delete it, then show the error in PRIu64 like below.

[image]

the same project and same files, i can't use clang to build the idf project which include the espressif__cbor component, it will show the macro error in PRIu64

[image]

[image]

### Steps to reproduce.

1. create v5.4 project
2. add espressif__cbor component
3. build in esp-clang
...

### Build or installation Logs.

```plain
<command line>:9:9: error: undefining builtin macro [-Werror,-Wbuiltin-macro-redefined]
    9 | #undef  __STDC_HOSTED__

After delete target_compile_options(${COMPONENT_LIB} PRIVATE "-U __STDC_HOSTED__"):

/home/gtc/workspace/S3project/managed_components/espressif__cbor/tinycbor/src/cbortojson.c:356:36: error: expected ')'
  356 |         if (fprintf(out, "{\"tag%" PRIu64 "\":", tag) < 0)
      |                                    ^
/home/gtc/workspace/S3project/managed_components/espressif__cbor/tinycbor/src/cbortojson.c:356:20: note: to match this '('
  356 |         if (fprintf(out, "{\"tag%" PRIu64 "\":", tag) < 0)
      |                    ^
/home/gtc/workspace/S3project/managed_components/espressif__cbor/tinycbor/src/cbortojson.c:364:40: error: expected ')'
  364 |             if (fprintf(out, ",\"tag%" PRIu64 "$cbor\":{", tag) < 0 ||
      |                                        ^
/home/gtc/workspace/S3project/managed_components/espressif__cbor/tinycbor/src/cbortojson.c:364:24: note: to match this '('
  364 |             if (fprintf(out, ",\"tag%" PRIu64 "$cbor\":{", tag) < 0 ||
      |                        ^
/home/gtc/workspace/S3project/managed_components/espressif__cbor/tinycbor/src/cbortojson.c:423:21: error: call to undeclared function 'cbor_value_to_pretty_advance'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  423 |     CborError err = cbor_value_to_pretty_advance(memstream, it);
      |                     ^
/home/gtc/workspace/S3project/managed_components/espressif__cbor/tinycbor/src/cbortojson.c:423:21: note: did you mean 'cbor_value_to_json_advance'?
/home/gtc/workspace/S3project/managed_components/espressif__cbor/tinycbor/src/cborjson.h:50:20: note: 'cbor_value_to_json_advance' declared here
   50 | CBOR_API CborError cbor_value_to_json_advance(FILE *out, CborValue *value, int flags);
      |                    ^
/home/gtc/workspace/S3project/managed_components/espressif__cbor/tinycbor/src/cbortojson.c:642:40: error: expected ')'
  642 |                 r = fprintf(out, "%s%" PRIu64, val < 0 ? "-" : "", ival);
      |                                        ^
/home/gtc/workspace/S3project/managed_components/espressif__cbor/tinycbor/src/cbortojson.c:642:28: note: to match this '('
  642 |                 r = fprintf(out, "%s%" PRIu64, val < 0 ? "-" : "", ival);
```

NewNoob2002, 2025-03-17:
@igrr hello, is this bug for cbor component?

NewNoob2002, 2025-03-17:
when i not add cbor component, i can use esp-clang build esp-idf-v5.4 project successfully

igrr, 2025-03-17:
I think it's an issue with Cbor component, yes. It probably worked with Clang some time in the past, but since we don't run CI builds with Clang for idf-extra-components, this use case got broken.

igrr, 2025-03-17:
cc @hmalpani

NewNoob2002, 2025-03-17:
ok, thanks, is it possible to fix this? I'd really appreciate it if so, as I'm currently learning to compile projects using llvm and clangd.

hrushikesh430, 2025-04-09:
Hello @NewNoob2002, Cloud you please share your minimal build application or the patch so that it would be easy for me to debug the code.

igrr, 2025-04-09:
@hrushikesh430 you can reproduce it like this (tested on the current master branch):

```shell
idf.py create-project testcbor
cd testcbor
export IDF_TOOLCHAIN=clang
idf.py add-dependency espressif/cbor
idf.py build
```

NewNoob2002, 2025-04-09:
> Hello [@NewNoob2002](https://github.com/NewNoob2002), Cloud you please share your minimal build application or the patch so that it would be easy for me to debug the code.您好，云，请分享您的最小构建应用程序或补丁，以便我轻松调试代码。

yes, Thank you for reply, you just need add this component in ESP-IDF component Registry, https://components.espressif.com/, then install esp-clang

hrushikesh430, 2025-05-19:
Hi @NewNoob2002, Thanks for reporting this issue, I am able to successfully reproduce your issue. I will raise the fix for this. Till then, you can fix this by removing `target_compile_options(${COMPONENT_LIB} PRIVATE "-U __STDC_HOSTED__")`. Let me know if this works for you.

NewNoob2002, 2025-05-21:
> Hi [@NewNoob2002](https://github.com/NewNoob2002), Thanks for reporting this issue, I am able to successfully reproduce your issue. I will raise the fix for this. Removing this line `target_compile_options(${COMPONENT_LIB} PRIVATE "-U __STDC_HOSTED__")` on my side is sufficient for me . Let me know if this works for you.您好，感谢您报告此问题，我成功复现了您的问题。我会提交修复方案。在我这边删除这行代码 `target_compile_options(${COMPONENT_LIB} PRIVATE "-U __STDC_HOSTED__")` 就足够了。如果您能成功，请告诉我。

Thank for reply, i try this but can't work, it show too many errors to stop, when you delete ths CmakeLists, it works?

[image]

hrushikesh430, 2025-05-21:
Hi @NewNoob2002, by deleting it works on my side. try giving `"-D__STDC_FORMAT_MACROS"` flag. Like `target_compile_options(${COMPONENT_LIB} PRIVATE "-D__STDC_FORMAT_MACROS")` and then check.

mahavirj, 2025-05-21:
@NewNoob2002 

Looks like your build could proceed further and there is a new error originating from [littlefs](https://github.com/joltwallet/esp_littlefs) component. This should be fixed separately or you can maybe add `-Wno-format` cflag to your project as an exception.

hrushikesh430, 2025-06-02:
Hi @NewNoob2002, closing this issue due to no response. Please feel free to reopen the issue for any query. Thank you.

NewNoob2002, 2025-06-03:
> Hi [@NewNoob2002](https://github.com/NewNoob2002), closing this issue due to no response. Please feel free to reopen the issue for any query. Thank you.

hello, i have tried to add target_compile_options(${COMPONENT_LIB} PRIVATE "-D__STDC_FORMAT_MACROS")
there show new format error in ESP_LOG,

[image]

[image]

closed 2026-09-28 by mahavirj