espressif/idf-extra-components #491: Controlling 6 LED strips

KacperStib, 2025-04-02:
### Answers checklist.

- [x] I have read the documentation of the component in question and the issue is not addressed there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
I am trying to use 6 WS2812B strips but when trying to create "led_strip_new_rmt_device()" it causes en error. I know that in ESP32 C3 there are only 2 RMT TX channels. Is there any way to share RMT channels along strips ? 
The only solution i found is "led_strip_del()" before starting to animate other strip, but it doesn't satisfy my program needs.

suda-morris, 2025-04-07:
Hi @KacperStib Could you expand what issue you met when you call the delete function?

> The only solution i found is "led_strip_del()" before starting to animate other strip, but it doesn't satisfy my program needs.

KacperStib, 2025-04-07:
> Hi [@KacperStib](https://github.com/KacperStib) Could you expand what issue you met when you call the delete function?
> 
> > The only solution i found is "led_strip_del()" before starting to animate other strip, but it doesn't satisfy my program needs.

Hello @suda-morris,
The solution with led_strip_del() is bad for me because, the strip loses it's former pixels colours. 
For steady backlight with same colour for all pixels in the strip this solution is OK. But I try to do some more advanced animation on this strip like rainbow chase. I simply animate in every step one pixel after antoher, so I need the memory, which is erased with calling "led_strip_del()" function.

suda-morris, 2025-04-08:
Hi @KacperStib Thanks for the feedback. It sounds reasonable to me. We will provide a way to dynamically switch the GPIO used by RMT TX channel.

zackees, 2025-04-28:
I've detached the leds from the api object and confirmed it works great. The array variable at the end of the struct needs to be converted to a pointer and everything just works.