espressif/idf-extra-components #497: ESP-IDF 5.3: adding this repository as a "component" triggers failures

SB211, 2025-05-04:
### Answers checklist.

- [x] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [x] I am using target and esp-idf version as defined in component's idf_component.yml
- [x] I have searched the [issue tracker](https://github.com/espressif/idf-extra-components/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

bdc_motor

### ESP-IDF version.

v5.3.3

### Development Kit.

ESP32

### Used Component version.

onewire_bus v1.0.2

### More Information.

Base OS is FreeBSD, recent ESP-IDF 5.3.3. espressif/idf-extra-components is aquired as git submodule and added to CMakeLists.txt via EXTRA_COMPONENT_DIRS.
When trying to build a small project having the aformentioned repo in path, preparation of the project always bails out, like

[...]
espressif/idf-extra-components/cbor/tinycbor/src/cbor.h:41:10: fatal error: tinycbor-export.h: No such file or directory
   41 | #include "tinycbor-export.h"
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
[...]

Similar error occurs in subfolder libsodium, where CMakeLists.txt incorporates a file "obsolete.c" which doesn't exist. By commenting that mistake out, the above error occurs, so I guess there is more to refurbish for ESP-IDF > 5.3 conformity.

I'm only interested in the subfolder for onewire_bus.

SB211, 2025-05-04:
Addendum: I assume the whole repository should function as a component.
A temporary workaround is to checkout the repo as such and do a symbolic link of subfolder "onewire_bus" located in folder components of the project.

igrr, 2025-05-04:
I think you are hitting these issues because the repository hasn't been cloned with the `--recursive` flag. Therefore all of the submodules aren't initialized.

You may either run `git submodule update --init` in the root directory of idf-extra-components, or add idf-extra-components/onewire_bus to EXTRA_COMPONENT_DIRS instead of the whole idf-extra-components.