espressif/idf-extra-components #500: Adding libcoap as a dependancy triggers a compile error

hani-n, 2025-05-08:
### Answers checklist.

- [x] I have read the component documentation [ESP-IDF Components](https://components.espressif.com) and the issue is not addressed there.
- [x] I am using target and esp-idf version as defined in component's idf_component.yml
- [x] I have searched the [issue tracker](https://github.com/espressif/idf-extra-components/issues?q=is%3Aissue) for a similar issue and not found any related issue.

### Which component are you using? If you choose Other, provide details in More Information.

coap

### ESP-IDF version.

release/v5.4

### Development Kit.

Custom board

### Used Component version.

espressif/coap

### More Information.

I am using esp-idf `release/v5.4` to build an application that requires `libcoap`, added as a dependancy from `idf.py add-dependancy coap`
The compile fails with unkown name types for `sockaddr_in` struct memebers i.e 
```
managed_components/espressif__coap/port/include/coap3/coap.h:29:5: error: unknown type name 'sa_family_t'
   29 |     sa_family_t     sin_family;
```
The issue seems to be related to the order in which `port/include/coap3/coap.h` declares variables/header files, more precisely the fact that structs are declared before header file inclusions. Declaring those structs afer header inclusions seems to resolve the issue.

mahavirj, 2025-05-13:
CC @mrdeep1

mrdeep1, 2025-05-13:
I'm unable to reproduce this issue.  Have you enabled the LwIP stack?

It would be good to know what the file was that included `coap.h`.  Was it one of your application files, or an internal libcoap build file?

However, I can see that in `port/include/coap3/coap.h` it includes `coap3/libcoap.h` which includes `sys/socket.h` which then includes `lwip/sockets.h` which is where `sa_family_t` is defined.

Please try making this change to see if things then work for you:
````
diff --git a/coap/port/include/coap3/coap.h b/coap/port/include/coap3/coap.h
index 59bf433..986a375 100644
--- a/coap/port/include/coap3/coap.h
+++ b/coap/port/include/coap3/coap.h
@@ -21,6 +21,8 @@
 extern "C" {
 #endif

+#include "coap3/libcoap.h"
+
 #if LWIP_IPV4
 #define COAP_IPV4_SUPPORT 1
 #else /* ! _LWIP_IPV4 */
@@ -48,8 +50,6 @@ struct sockaddr_in6 {
 #endif /* INET6_ADDRSTRLEN */
 #endif /* ! LWIP_IPV6 */

-#include "coap3/libcoap.h"
-
 #include "coap3/coap_forward_decls.h"
 #include "coap3/coap_address.h"
 #include "coap3/coap_async.h"
````
and revert back to us.

hani-n, 2025-05-13:
@mrdeep1 This change does resolve the issue !

hani-n, 2025-05-13:
It was a `coap_handlers` header file in the application code

mrdeep1, 2025-05-13:
OK.  If  you include `coap3/libcoap.h` in your application code header file before `coap3/coap.h`, does that work instead of this alternative fix?

hani-n, 2025-05-13:
It does not, I would also need to include `coap_session_internal` since I am using `coap_session_t` and other structs.
That being said, the initial compile issues remain.

mrdeep1, 2025-05-16:
The `coap_session_t *session` parameter is a pointer to an opaque structure, and there should be no compilation issues (other than perhaps an unused variable warning) providing you have defined '#include <coap.h>' prior to the function, or to any functions that pass in the function to internal libcoap code.

[coap_session(3)](https://libcoap.net/doc/reference/develop/man_coap_session.html) provides a set of function definitions that can be used to access variables within the `coap_session_t`.  A common usage here is [coap_session_get_contex(3)](https://libcoap.net/doc/reference/develop/man_coap_session_get_context.html#coap_session_get_context)()